TheThingsNetwork/lorawan-stack #4564: Cancelling The Things ID login leads to "Forbidden" error

nejraselimovic, 2021-08-31:
#### Summary
Cancelling login via The Things ID leads to "Forbidden" error

#### What do you see now?
[image]
[image]

#### What do you want to see instead?
Instead of this error, a user should probably be redirected to the login page

#### Environment
TTS Community Edition

#### Can you do this yourself and submit a Pull Request?
@kschiffer?

kschiffer, 2021-09-15:
So I've spent some time looking into other services that use OpenID Connect (our auth flow in TTID) and from what I can see, Google, Facebook, and Apple do not even provide a cancel option when using their services to login. This makes me wonder if we even need such functionality when the user can just as well use the browser's back button.

Technically, the only difference I would see is that the state cookie would remain stored but I don't believe this is problematic, since every newly initiated login would invalidate the cookie anyway.

UX-wise I don't believe removing the cancel button would be problematic in any way.

kschiffer, 2021-09-16:
I'd like to open this up for discussion. @htdvisser @bafonins @NicolasMrad

NicolasMrad, 2021-09-17:
I just visited random websites to check the log-in process and none of them have a cancel button, so in my opinion users presses or going back is sufficient without having a cancel button.

htdvisser, 2021-09-17:
I don't think a cancel button makes sense when a user actively tries to login. However, if a user gets automatically redirected to TTID, I think the UI should show them that they have a choice between continuing and cancelling. Especially if pressing "back" simply redirects them to the login page again, I think it's better to explicitly tell the OAuth client "no".

kschiffer, 2021-09-20:
What should happen then, upon clicking cancel?

kschiffer, 2021-09-20:
See also https://discuss.thethingsindustries.com/t/ttsce-and-ttid-login-ux/655

htdvisser, 2021-09-28:
Conclusion after discussion: we just need to catch the error, show a nicer message and a button to the solution (go to login page)

PR #4849 (2021-11-10): Provide better error when OIDC login is aborted
#### Summary
Closes #4564 

#### Changes

- Check for OIDC abort error and display a better error message
- Provide a login button also for errors happening during the oidc callback (was regular callback only)

#### Testing

Manual testing

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentatio [... cut]