TheThingsNetwork/lorawan-stack #4630: Pick up LoRa Cloud GNSS data from payload formatter field

benolayinka, 2021-09-15:
#### Summary

We should ask device makers to break out Semtech GNSS data for the LR1110 in a payload formatter, and then pick it up in the LoRa Cloud integration by looking for that key

#### Why do we need this?

- The GNSS data does not come in uplinks in a standardized form. On the yabby edge, for example, it is contained in variable length wifi/gnss uplinks as the last 8 bytes. This needs to be defined in a payload formatter
- By looking for that key, we can stop doing unnecessary LoRa Cloud requests for devices with no GNSS data (currently, enabling the LoRa Cloud integration creates an error on every uplink)

#### What is already there? What do you see now?

When the LoRa Cloud integration is enabled for an application, all uplinks are forwarded to LoRa Cloud regardless of whether they contain valid gnss data

#### What is missing? What do you want to see?

Only make LoRa cloud requests based on the presence of a gnss data key in the decoded output

#### Can you do this yourself and submit a Pull Request?

@adriansmares

jpmeijers, 2021-09-20:
This sounds like the type of logic that could be offloaded to the application payload parser:
https://github.com/TheThingsNetwork/lorawan-application-payload

benolayinka, 2021-09-23:
> This sounds like the type of logic that could be offloaded to the application payload parser:
> https://github.com/TheThingsNetwork/lorawan-application-payload

What is it? Docs dont say

johanstokking, 2021-10-06:
OK, I didn't realize this is Geolocation and not DAS, thanks for clearing that up.

> and if a GNSS-like field is found to send that to GLS (instead of the whole frame)

And what if it's not there?

adriansmares, 2021-10-06:
> And what if it's not there?
I would argue that the GLS request should not be done - then the order is:
- If no decoded payload is available, send the full frame.
- If the decoded payload is available, and there is no GNSS payload, do nothing.
- If the decoded payload is available, and there is a GNSS payload, send that.

johanstokking, 2021-10-06:
OK. And I'm presuming that this aligns with the WiFi lookup, right?

adriansmares, 2021-10-06:
> OK. And I'm presuming that this aligns with the WiFi lookup, right?

Indeed, and maybe we can move that to https://github.com/TheThingsNetwork/lorawan-application-payload too.

htdvisser, 2022-01-05:
Doesn't look like this is going to get finished for v3.17.0, so pushing to v3.17.1.

@NicolasMrad: This has been pushed forward since October, so I think we should re-evaluate the priority.

closed 2022-01-25 by adriansmares