TheThingsNetwork/lorawan-stack #4650: Allow fetching profile information using OAuth token

romach, 2021-09-21:
#### Summary

Now we can get profile information on https://eu1.cloud.thethings.network/oauth/profile-settings page.

But we can get it only using browser.

It will be useful to be able to fetch this information using OAuth 2.0 flow.

#### Why do we need this?

In many API calls we need **user id** but now we can get it from https://eu1.cloud.thethings.network/oauth/profile-settings page. It will be useful to allow to get profile information using Oauth flow.

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

kschiffer, 2021-09-23:
If you already obtained an access token via the OAuth2 flow, you can use the `auth-info` endpoint of the API to get information about the authenticated user:

`GET /api/v3/auth_info` made with the Bearer token (your access token) in the Authorization header would yield:

```json
{
  "oauth_access_token": {
    "user_ids": {
      "user_id": "kschiffer"
    },
    …
}
```

kschiffer, 2021-10-04:
Closing this for now. @romach please reopen if this does not resolve your problem.

closed 2021-10-04 by kschiffer