TheThingsNetwork/lorawan-stack #4652: Removed obsolete last invalidation keys

adriansmares, 2021-09-21:
#### Summary

https://github.com/TheThingsNetwork/lorawan-stack/pull/4651 removes the `ttn:v3:ns:application-uplinks:uid:<uid here>:last-invalidation` keys from the application uplink queue. These keys are now unused, and may be deleted.

#### Why do we need this?

In order to avoid storing unused keys.

#### What is already there? What do you see now?

Keys of the form `ttn:v3:ns:application-uplinks:uid:*:last-invalidation`

#### What is missing? What do you want to see?

These keys should be deleted.

#### Environment

`v3.15.0`

#### How do you propose to implement this?

`ns-db` migrate takes care of the `v3.10` -> `v3.11` migration. This migration needs to be extracted in a separate function / file, and a new one that ranges through the invalidation keys and deletes them should be added.

#### How do you propose to test this?

Manually set the key and make sure that the migration deletes it safely.

#### Can you do this yourself and submit a Pull Request?

Yes.

htdvisser, 2022-01-05:
Doesn't look like this is going to get finished for v3.17.0, so pushing to v3.17.1.

NicolasMrad, 2023-01-02:
This is created a while ago, can you advise on the importance of this?

adriansmares, 2023-01-18:
I would say this is still relevant and that we can write a short migration for this. Assigning @cvetkovski98 to take a look into this for the next minor - let me know if you need extra details.

PR #6043 (2023-02-14): Add last invalidation keys removal migration
#### Summary
Replace the v3.10 to v3.11 Network Server db migration with last invalidation keys and add version warnings accordingly.

Closes #4652 

#### Changes
- Removed the v3.10 to v3.11 key migration from `ns-db migrate`
- Added last invalidation keys migration

#### Testing
- Added the keys to Redis and verified that they don't exist after the migration
- Verified that the version numbers are properly set after the migration under the correct namespace 

##### Regressions
- Direct migrations of version 0 to version 2 of the Network Server db  [... cut]
closed 2023-03-07 by NicolasMrad