TheThingsNetwork/lorawan-stack #4685: Avoid consecutive disconnect/reconnect messages in the Live Data view

kschiffer, 2021-10-01:
#### Summary
The live data view currently shows many consecutive connect and reconnect events when the internet connection is lost. This is frequently the case when e.g. closing the laptop or leaving the browser tab unfocused for a long time. We should combine such messages so that only two are shown.

[image]

#### Steps to Reproduce

1. Go to live data view
2. Either leave the browser tab unfocused for a long while or (easier) disconnect from Wi-Fi or internet a couple of times
3. See many synthetic events being generated about that consecutively for each disconnect event

#### What do you see now?
Many disconnect/reconnect errors being generated consecutively.

#### What do you want to see instead?
Such consecutive events to be combined into just two events:
1. The first disconnect event of the consecutive group
2. The last reconnect event of the consecutive group

This also means that such events should not be combined if there was an actual stream event in between them.

#### How do you propose to implement this?
It's probably best to solve this at the event reducer level, by changing the reducer to only add the synthetic reconnect/disconnect event if there haven't been the same events immediately preceding. In that case, we can simply remove the last reconnect event and replace it with the more recent one.

#### How do you propose to test this?
Manual testing will suffice for now.

#### Can you do this yourself and submit a Pull Request?
This is a good first issue.

bafonins, 2021-12-23:
@ryaplots Can we close this issue?

ryaplots, 2022-05-24:
Still can't reproduce this

KrishnaIyer, 2024-10-14:
This is still relevant as I see it in the current version of the console. Can we think of ways to hide the multiple "disconnected"/ "reconnected" loops and have better UX for this?

[image]

KrishnaIyer, 2024-12-02:
Did a quick check with turning off WiFi and switching SSIDs and this works pretty well.
[image]
[image]
Let's add to this after the RFT if needed.

closed 2024-12-02 by KrishnaIyer