TheThingsNetwork/lorawan-stack #4686: Internal Error reported

niclash, 2021-10-01:
#### Summary
Unable to create an end-device from command line tool `ttn-lw-cli`. I am using the free community cloud.

I am not sure if this is an internal server error as reported, or that it is not providing a meaningful error message about what I have done wrong.

```
ttn-lw-cli end-device create link2web-dummy2 room1 --dev-eui=a84041287182a048 --join-eui=a000000000000100 --frequency-plan-id=EU_863_870_TTN --formatters.up-formatter=FORMATTER_JAVASCRIPT --formatters.up-formatter-parameter="function decodeUplink(input) {
  var bytes = input.bytes;
  return {
    data: {
      bytes: input.bytes,
      sensor: 'LHT65',
      battery:((bytes[0]<<8 | bytes[1]) & 0x3FFF)/1000,
      temperature_SHT:((bytes[2]<<24>>16 | bytes[3])/100).toFixed(2),
      humidity:((bytes[4]<<8 | bytes[5])/10).toFixed(1),
      temperature_DS: ((bytes[7]<<24>>16 | bytes[8])/100).toFixed(2)
    },
    warnings: [],
    errors: []
  };
}"
```

results in 
```
  WARN Finished unary call                      duration=63.9ms error=rpc error: code = Internal desc = error:pkg/rpcserver:rpc_recovered (Internal Server Error) error_correlation_id=fdef690b0aec41e89e18a8b49ad9783a error_name=rpc_recovered error_namespace=pkg/rpcserver grpc.method=Set grpc.service=ttn.lorawan.v3.NsEndDeviceRegistry grpc_code=Internal namespace=grpc
 ERROR Could not create end device, rolling back... error=error:pkg/rpcserver:rpc_recovered (Internal Server Error)
 ERROR Could not delete end device from Application Server device_uid=link2web-dummy2.room1@ttn error=error:pkg/applicationserver:device_not_found (device `link2web-dummy2.room1@ttn` not found)
 ERROR Could not delete end device from Network Server error=error:pkg/networkserver:device_not_found (device not found)
error:pkg/rpcserver:rpc_recovered (Internal Server Error)
    correlation_id=fdef690b0aec41e89e18a8b49ad9783a
```

niclash, 2021-10-01:
I seem to witness an issue with `--frequency-plan-id=EU_863_870_TTN`. If I drop it, it goes through.

niclash, 2021-10-01:
So, for future readers; One needs to provide "everything" correctly or there is a great probability that this shows up.

For TTN developers; More informative error message would be nice.

Example of working command;
```
ttn-lw-cli end-device create link2web-dummy9 room3 --dev-eui=a840417221822fb5 --join-eui=a8404169e1822fb5 --root-keys.app-key.key=6f686fcbbff64c12673fd161be93c365 --lorawan-version=MAC_V1_0_2 --lorawan-phy-version="PHY_V1_0_3_REV_A" --network-server-address=eu1.cloud.thethings.network --frequency-plan-id=EU_863_870_TTN --supports-join=true --formatters.up-formatter=FORMATTER_JAVASCRIPT --formatters.up-formatter-parameter="function decodeUplink(input) {
  var bytes = input.bytes;
  return {
    data: {
      bytes: input.bytes,
      sensor: 'LHT65',
      battery:((bytes[0]<<8 | bytes[1]) & 0x3FFF)/1000,
      temperature_SHT:((bytes[2]<<24>>16 | bytes[3])/100).toFixed(2),
      humidity:((bytes[4]<<8 | bytes[5])/10).toFixed(1),
      temperature_DS: ((bytes[7]<<24>>16 | bytes[8])/100).toFixed(2)
    },
    warnings: [],
    errors: []
  };
}"
```

htdvisser, 2021-10-04:
What do you expect me to do here, @kschiffer?

kschiffer, 2021-10-04:
Evaluate whether a 

> More informative error message

is required and otherwise close this issue.

htdvisser, 2021-10-05:
I was able to reproduce the error. Creating an end device in the Network Server can fail if you supply a frequency plan, but not the PHY version -- [details in Sentry](https://sentry.io/organizations/the-things-industries/issues/2470206875/?project=2732897) (TTI only).

For this specific error, there isn't much more information we can give to end users here. It was an internal server error, and at best, we could have returned _"runtime error: invalid memory address or nil pointer dereference"_ which wouldn't be very helpful to end users.

We'll track the actual bug elsewhere, there's nothing to do for this issue.

closed 2021-10-05 by htdvisser