TheThingsNetwork/lorawan-stack #4697: Add Status Page references in the TTS console

ymgupta, 2021-10-02:
#### Summary
We have status pages for the operational status of The Things Stack deployments (TTS Cloud & TTS Community Edition) and Packet Broker. Mention these references in the TTS console so that users will be aware and easily redirected to the respective status pages to know the status in case of any incidents.
- TTS Cloud: https://status.thethings.industries/
- TTS Community Edition: https://status.thethings.network/
- Packet Broker: https://status.packetbroker.net/

#### Why do we need this?
Users will be aware of the current operational status of the product.

#### What is already there? What do you see now?
References to the Documentation and Release notes at the footer section of the console.

#### What is missing? What do you want to see?
Status page references in the TTS console.
**Optional**: It would be good if we can add a configuration option in TTS to set the reference of the status page, such that self-hosted users can set the reference of their status page. 
cc: @michalborkowski96 

#### Environment
TTS V3.15.0
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No @kschiffer

michalborkowski96, 2022-01-06:
@ryaplots When this gets done, please create an issue in [`lorawan-stack-aws`](https://github.com/TheThingsIndustries/lorawan-stack-aws) so that I can add relevant config option to the AWS templates

ryaplots, 2022-03-28:
> @ryaplots When this gets done, please create an issue in [`lorawan-stack-aws`](https://github.com/TheThingsIndustries/lorawan-stack-aws) so that I can add relevant config option to the AWS templates

@michalborkowski96 Here you have the open issue: [#623](https://github.com/TheThingsIndustries/lorawan-stack-aws/issues/623)

closed 2022-04-11 by ryaplots