TheThingsNetwork/lorawan-stack #4698: Annoying web UI bug: toolbar moving left/right very quickly

sandorspruit, 2021-10-02:
#### Summary
The logo with the cloud and "The Things network" in the top left corner of the web UI seems to exist in two slightly different sizes. One has the text "network" in a slightly larger font. For some reason, they get swapped in and out, typically as I move my mouse across the page. As one image is a couple of pixels larger than the other, this causes a rather annoying distracting shiver in the toolbar at the top of the page.. 

#### Steps to Reproduce

1.  Enter the console web UI
2. Move the mouse over the toolbar, e.g. to select "Applications", then a particular Application 
3. Move the mouse away from the toolbar, e.g. to inspect properties of an Application
4. Logo in the top left corner seems to get replaced, maybe in mouse event handler

#### What do you see now?

(not 100% of the time) Toolbar shivers as larger logo gets swapped in and out

#### What do you want to see instead?

One logo, one size, toolbar stays in place.

#### Environment

MacOS Big Sur/Safari 15

#### How do you propose to implement this?

I'd check mouse handler code, or check the images in web resources for any similar images of difference sizes

#### How do you propose to test this?

I see this happening so often, although not 100% of the time, that moving my mouse around in the interface would suffice to check whether the issue is resolved.

#### Can you do this yourself and submit a Pull Request?

Might give it a try, although I am not sure I am familiar with the software used.

sandorspruit, 2021-10-02:
[image]

sandorspruit, 2021-10-02:
As the image is a single SVG, the problem can't be related to an image getting updated on-mouse-over. I do see an error in the Safari developer tools about a specific font missing: https://assets.cloud.thethings.network/source-sans-pro-v13-latin_latin-ext-700.e7cfe2ca0e4a2f2caf7bfb2f11028e92.woff2. This is the larger of the fonts in the assets folder, so maybe the SVG gets rendered with the smaller font of the same name?

sandorspruit, 2021-10-02:
Could very well be a Safari bug that I stumbled upon just now, e.g. something like this looks similar https://stackoverflow.com/questions/19215931/displaying-custom-fonts-in-svg-for-ios-chrome-and-safari

kschiffer, 2021-10-11:
Thanks for filing this issue @sandorspruit

I've applied a fix for this issue and it will be resolved in the community network in the next deployment. Closing this as such.

closed 2021-10-11 by kschiffer