TheThingsNetwork/lorawan-stack #4816: Gateway fails to get gps_sync

virtualguy, 2021-11-03:
#### Summary
Using a Tektelic Kona Macro with the udp packet forwarder the gateway doesn't syncronise absolute time although the gps is running. It appears the gateway sends the `tmms` field but not a `time` field.

This leads to being unable to send class b messages due to no gps_sync

#### Steps to Reproduce
1. Connect a tektelic gateway with gps to TTS
2. Send uplinks, optionally from a device configured for class b

#### What do you see now?
From logs gateway does not reach absolute time sync
`DEBUG	Synchronized server absolute time only`

From console class b messages fail to send

#### What do you want to see instead?
Gateway reaches gps time sync
`DEBUG	Synchronized server and gateway absolute time`

#### Environment
tts v3.15.3

Tektelic packet forwarder 5.0.2

Gateway log:
```
JSON up: {"stat":{"time":"2021-11-03 10:27:59 GMT","lati":-36.8XXXX,"long":174.9XXXX,"alti":60,"rxnb":3,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1,"temp":32,"cpur":37.8,"memr":16.0}}

...

JSON up: {"rxpk":[{"tmst":474859052,"tmms":1319970519707,"chan":0,"rfch":0,"freq":916.800000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":11.8,"rssi":-39,"size":104,"data":"gMwRUwAQDQABIIejuf30FVvhQ8vyhA7AT9PeOBpWT2efMSzfj3PxiFEtF+CVeHzxIMVpi0ctniNv9SC2+Mbo8KGz4Qb4EsYOPuR6OwBDciawBK86HE/7dXW43IcUqA8jyl6Lfsx0lPY="}]}
```

Class B downlinks fail with the following error: pkg/gatewayserver/io no_gps_sync
[send-failure.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/7466909/send-failure.txt)
#### How do you propose to implement this?
Use the tmms field for time sync if present

#### How do you propose to test this?
Can test with gateways and dev stack

#### Can you do this yourself and submit a Pull Request?
Need help, gatewayserver owners: @johanstokking @KrishnaIyer @adriansmares?

adriansmares, 2021-11-03:
I guess this diff could probably fix it:

```diff
diff --git a/pkg/ttnpb/udp/translation.go b/pkg/ttnpb/udp/translation.go
index 673ead2dba..a8ad009c89 100644
--- a/pkg/ttnpb/udp/translation.go
+++ b/pkg/ttnpb/udp/translation.go
@@ -193,8 +193,14 @@ func convertUplink(rx RxPacket, md UpstreamMetadata) (ttnpb.UplinkMessage, error
                }
        }
 
-       if rx.Time != nil {
-               goTime := time.Time(*rx.Time)
+       var goTime time.Time
+       switch {
+       case rx.Time != nil:
+               goTime = time.Time(*rx.Time)
+       case rx.Tmms != nil:
+               goTime = gpstime.Parse(time.Duration(*rx.Tmms) * time.Millisecond)
+       }
+       if !goTime.IsZero() {
                for _, md := range up.RxMetadata {
                        md.Time = &goTime
                }
```

The problem is that I don't actually know what the protocol says about this, I [see](https://github.com/Lora-net/packet_forwarder/blob/d0226eae6e7b6bbaec6117d0d2372bf17819c438/PROTOCOL.TXT#L442-L444) only that they added the field, not that they are no longer backfilling `time`. @johanstokking / @KrishnaIyer any idea if this is conforming behavior ?

johanstokking, 2021-11-03:
We should use `tmms` only.

Using `time` for absolute time is problematic. As you see in @virtualguy's example, sometimes there's no second fraction and sometimes there is (see reference implementation snippet that @adriansmares referred to). Second precision isn't enough for class B, so if anything, we should prefer `tmms` over `time`, but I think we should only consider `tmms`.

adriansmares, 2021-11-03:
The first message is a status message, and we don't do clock sync with those, so I think giving priority to `tmms` over `time` should be adequate (instead of just `tmms`).

closed 2021-11-03 by adriansmares