TheThingsNetwork/lorawan-stack #4823: UDP Gateway drops gps sync on tx event

virtualguy, 2021-11-04:
#### Summary
When the scheduler sends a tx event for a gateway it attempts a time sync using the uplink token.

This clock sync sets the gateway time to nil and then stops class b downlinks until the next uplink.

For busy gateways this may not be evident as they may recieive another uplink before the RX1/RX2 time

#### Steps to Reproduce
1. Connect gateway with gps
2. Run a single node in class b mode

#### What do you see now?
Class B Downlinks fail with `no_gps_sync`

[gps-sync-dropout.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/7473984/gps-sync-dropout.txt)

#### What do you want to see instead?
Absolute time sync stays set and class b traffic is successful

#### Environment

v3.16-dev (1c1a21dbc53d21e9e480dbbdff46aa0f3aa8eb47)

#### How do you propose to implement this?
Quick fix was to hold off a clock sync of the last sync was within 30 seconds. 

```
diff --git a/pkg/gatewayserver/scheduling/scheduler.go b/pkg/gatewayserver/scheduling/scheduler.go
index 4a044ccf6..bdf0023a8 100644
--- a/pkg/gatewayserver/scheduling/scheduler.go
+++ b/pkg/gatewayserver/scheduling/scheduler.go
@@ -254,7 +254,7 @@ func (s *Scheduler) syncWithUplinkToken(token *ttnpb.UplinkToken) bool {
        if token.GetServerTime().IsZero() || token.GetConcentratorTime() == 0 {
                return false
        }
-       if lastSync, ok := s.clock.SyncTime(); ok && lastSync.After(token.ServerTime) {
+       if lastSync, ok := s.clock.SyncTime(); ok && lastSync.Add(time.Second*30).After(token.ServerTime) {
                return false
        }
        s.clock.SyncWithGatewayConcentrator(token.Timestamp, token.ServerTime, ConcentratorTime(token.ConcentratorTime))
```

A more robust approach would be to observe the gateway time if it was set on the uplink and then not attempt another sync

#### How do you propose to test this?
Can test patches on existing hardware

#### Can you do this yourself and submit a Pull Request?
@johanstokking @KrishnaIyer @adriansmares?

...

adriansmares, 2021-11-04:
I guess the uplink token should contain the absolute gateway time (`time`/`tmms`) in order to allow synchronization with absolute time via the uplink token.

johanstokking, 2021-11-04:
> I guess the uplink token should contain the absolute gateway time (`time`/`tmms`) in order to allow synchronization with absolute time via the uplink token.

Yes indeed

closed 2021-11-08 by adriansmares