TheThingsNetwork/lorawan-stack #4835: Make it easier to configure Basic Auth for webhook integrations in the Console

htdvisser, 2021-11-08:
#### Summary

We need to make it easier to configure Basic auth for webook integrations.

#### Why do we need this?

Basic Auth is a very common auth method for (self-built) integrations, so it would be nice if we could make the Console handle this as a special case. 

#### What is already there? What do you see now?

All headers are treated the same.

#### What is missing? What do you want to see?

The `Authorization` header should get special treatment for `Basic` auth.

#### How do you propose to implement this?

- If `headers` doesn't already have an `Authorization` key with a value that starts with `Basic`, show a button to add one.
- If `headers` does have one, base64-decode the value, split on the first `:` and display Username and Password fields.
- When the Username and/or Password fields change, update the header to the base64-encoded concatenation of Username, `:` and Password.

kschiffer, 2021-12-06:
I think this can be implemented just as @htdvisser has drafted it here.

Within the webhook form, below the header settings, we can add a button `Add Basic Auth header`. Once clicked it will compose the header as described.

kschiffer, 2022-01-04:
Here's a wireframe of the functionality:

[image]

The tooltip should read

> To increase access security, you can choose to generate a “basic auth” authorization header to be attached to the webhook requests, if the target server requires doing so. This will authenticate the webhook requests with the defined credentials.

PR #5077 (2022-02-21): Configure basic auth

#### Summary

This pull request closes #4835, by adding a basic auth configuration field to webhook form 

#### Changes

- Add basic auth configuration field
[image]
[image]

- Add warning when the user is trying to add another Authorization header
[image]

#### Testing

Manually

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target bra
closed 2022-02-21 by ryaplots