TheThingsNetwork/lorawan-stack #4841: Device Claiming Server

johanstokking, 2021-11-09:
#### Summary

Implement a Device Claiming Server.

#### Why do we need this?

To facilitate claiming of end devices.

#### What is already there? What do you see now?

We currently have a proprietary implementation of the Device Claiming Server. This issue replaces at least the end device claiming part and makes the DCS component groundwork open source.

#### What is missing? What do you want to see?

- Simple groundwork for the DCS component
- Configuration of JoinEUI prefix to a protocol, target address, credentials, etc
- Expose `EndDeviceClaimingServer` with claiming, unclaiming and getting claim info
- DCS needs to authenticate with Join Servers on behalf of the application collaborator

#### How do you propose to implement this?

1. Rework `EndDeviceClaimingServer` service:
    1. Remove application authorization; `AuthorizeApplication()` and `UnauthorizeApplication()` or simply return `Unimplemented`
    2. Report whether claiming is available for a given JoinEUI with new rpc `GetInfoByJoinEUI()`
    3. Claim an end device; existing rpc `Claim()` suffices, but most fields aren't relevant anymore as DCS won't deal with NS and AS registration. DCS will call out to the Join Server on behalf of the user and is the "application owner"
    4. Get claim status of an end device with new rpc `GetClaimStatus()`. This takes `EndDeviceIdentifiers`
    5. Unclaim with new rpc `Unclaim()`
2. Implement The Things Join Server and LoRa Cloud Join Server as client protocols

#### Can you do this yourself and submit a Pull Request?

We need to decide between:

1. Registering each TTS application with its own credentials at the Join Server. This means that TTS needs a way to manage device groups (applications) at the Join Server. Each device group would have its own auth. With LoRa Cloud Join Server, we can't programmatically create a new "application owner", so users would have to copy-paste their application owner TLS client cert and key so DCS can act on behalf of the user. This benefit of mapping TTS applications to JS device groups is that the JS can enforce authentication. The downside is that DCS needs to keep per-application credentials for each JS. Another downside is that customers are not able to manage their devices themselves in LoRa Cloud Join Server, unless of course they run The Things Stack themselves

2. Considering a TTS cluster with AS and DCS one "Application Server" at the Join Server. In LoRaWAN Backend Interfaces terms, this would be one ASID. In LoRa Cloud Join Server terms, a TTS cluster is an "application owner". The benefit is that this is easiest to configure. The downside is that TTS has to make sure that you can't unclaim devices that are not registered in the application (DCS would basically need to get the device from IS).

My preference is with (2).

@htdvisser @KrishnaIyer thoughts?

KrishnaIyer, 2021-11-10:
> Considering a TTS cluster with AS and DCS one "Application Server" at the Join Server. In LoRaWAN Backend Interfaces terms, this would be one ASID. In LoRa Cloud Join Server terms, a TTS cluster is [...]

This sounds good to me.

> Report whether claiming is available for a given JoinEUI with new rpc GetInfoByJoinEUI()
> Get claim status of an end device with new rpc GetClaimStatus(). This takes EndDeviceIdentifiers

In both these cases, what auth would the user provide and how to assign rights? If we allow anyone to query if a particular EUI is claimed or not, wouldn't we be allowing users to scan the claim status of every EUI possible? In Gateway claiming, we return "error already claimed" only after we validate the CAC for the EUI (but then we have the IS there).

johanstokking, 2021-11-11:
> In both these cases, what auth would the user provide and how to assign rights? If we allow anyone to query if a particular EUI is claimed or not, wouldn't we be allowing users to scan the claim [...]

The rpc `GetInfoByJoinEUI()` is only taking a JoinEUI. Whether claiming is supported by a JoinEUI is pretty much public, it will be in DNS in the future anyway.

The rpc `GetClaimStatus()` requires application rights. DCS takes `EndDeviceIdentifiers` and checks rights to the application. DCS then dials out to the Join Server to get the claim status by DevEUI.

htdvisser, 2021-11-11:
> Remove application authorization; AuthorizeApplication() and UnauthorizeApplication() or simply return Unimplemented

That's also what gRPC will return if we remove the implementations.

> The rpc GetInfoByJoinEUI() is only taking a JoinEUI. Whether claiming is supported by a JoinEUI is pretty much public, it will be in DNS in the future anyway.

We can apply the same logic as with some of our other "public" RPCs: accepting any valid auth token on the network.

> Considering a TTS cluster with AS and DCS one "Application Server" at the Join Server. [...] The downside is that TTS has to make sure that you can't unclaim devices that are not registered in the application (DCS would basically need to get the device from IS).

How would this work with self-hosted Open Source / Enterprise deployments and with Dedicated Cloud deployments? Wouldn't it make more sense to have only a single DCS for each network, deployed together with the IS?

johanstokking, 2021-11-11:
> How would this work with self-hosted Open Source / Enterprise deployments and with Dedicated Cloud deployments?

Per Backend Interfaces, Join Server associates an AS-ID with an end device. This is a string, fortunately not an EUI64 this time; I'd say for TTS we use cluster addresses as AS-IDs. 

> Wouldn't it make more sense to have only a single DCS for each network, deployed together with the IS?

I think it is valuable to keep AS-IDs separate across clusters, so the Join Servers can authenticate Application Servers separately. We do not have to provision multi-cluster deployments with the same credentials to authenticate as one AS-ID.

johanstokking, 2021-12-20:
@KrishnaIyer please see https://github.com/TheThingsIndustries/lorawan-join-server/pull/5.

The idea is that DCS gets a map of JoinEUI to client settings. The client settings contain the API to use as well as authentication methods. In case of our new JS, that will be the new claiming API and HTTP basic authentication settings. DCS authenticates as Application Server and needs to claim, unclaim, get claim status and update a claim on behalf of a user.

PR #5324 (2022-03-22): Add support for The Things Join Server

#### Summary

Closes #4841 

#### Changes

- Backport parts of https://github.com/TheThingsIndustries/lorawan-stack/pull/3073
- Simple file copy. Nothing added functionally.

#### Testing

NA

##### Regressions

NA

#### Notes for Reviewers

- Since this needs the TTJS to work, I will add documentation for the configuration only during/after we have TTJS deployed.
- Without the configuration, this will just use the fallback EDCS in enterprise and do nothing (as it is now) in OS.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are n [... cut]
closed 2022-03-22 by KrishnaIyer