TheThingsNetwork/lorawan-stack #4846: Console > Applications > Live data - must show best SNR/RSSI pair instead of only values from first gateway

lnlp, 2021-11-09:
#### Summary
In 'Console > Applications > _application_ > Live data' entries, SNR and RSSI values are displayed.
Currently these SNR and RSSI values are taken from the first gateway listed in the JSON data which makes no sense.

When an uplink is received by multiple gateways, always displaying SNR and RSSI values only from the first gateway makes no sense. From application perspective this is not representative for the signal quality of the uplink message received by the application.

One usually wants to know how well the uplink message was received. When the message was received by multiple gateways this means: the best SNR/RSSI values pair from any of the gateways that received that uplink.

The first gateway in the list is not the one with the best SNR/RSSI values (unless accidental).
The currently displayed SNR and RSSI values are not representative and hence practically useless (bug).

The best SNR/RSSI pair from all gateways that received the uplink shall be displayed instead.

#### Steps to Reproduce

Just check Console > Applications > Live data for uplink messages that were received by multiple gateways.
Compare SNR and RSSI values listed in the Data preview field with values in the JSON details.

#### What do you see now?

SNR and RSSI values from first gateway in the list.

#### What do you want to see instead?

SNR and RSSI values from the gateway with the best SNR/RSSI value pair.

#### Environment

Not relevant.

#### How do you propose to implement this?

Determine which gateway shows the best SNR/RSSI value pair and show those SNR and RSSI values instead.

The SNR and RSSI values should be from the same gateway of course.

I'm not sure how to compare SNR/RSSI combinations, but the (RF) experts will probably know how to determine ranking.

#### How do you propose to test this?

Just verify in the console.

#### Can you do this yourself and submit a Pull Request?

No.

kschiffer, 2022-01-26:
Thanks for filing this issue @lnlp!

I will schedule to fix this soon.

htdvisser, 2022-02-22:
We're not going to do that. Consumers should not rely on RX metadata being in any particular order. If consumers want to do some aggregation to find the min/avg/max SNR/RSSI/etc. they should do that themselves.

lnlp, 2022-02-24:
@kschiffer @ryaplots @htdvisser 

This issue was closed for what reason?, because already planned for/implemented in v3.18.1?

Or incorrectly closed because of "is it possible to sort ...", "we're not going to do that"?
The sorting is unrelated to the original issue.

The original issue should be fixed, otherwise SNR and RSSI values shown in the console are totally useless.

kschiffer, 2022-02-24:
For technical reasons, we have to resort to the highest SNR for the time being. The pair calculasion is not trivial, since RSSI is not a normalized measure and can vary depending on the gateway antenna. Also in the backend, we are currently mainly using the highest SNR to determine gateways for downlinks. This calculation will be updated at a later stage at which point we can reconsider the Consoles calculation as well.

> because already planned for/implemented in v3.18.1?

Yes. Keep in mind that merging does not mean that the feature is immediately active in our deployments, but will be updated as per our release and deployment schedule in the days after the release.

PR #5242 (2022-02-23): Get gateway with highest SNR

#### Summary

This PR closes #4846 

#### Changes

- Introduced a function to get the gateway that has the highest SNR to be displayed in the event preview.

#### Testing

Manually.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: Relevant documentation is added or updated.
- [x] Changelog: Significant  [... cut]