TheThingsNetwork/lorawan-stack #4850: Show "Last seen" when a gateway is offline

kschiffer, 2021-11-09:
#### Summary
For offline or disconnected gateways, the Console should display when they were last active (if applicable).

#### Why do we need this?
Better diagnosing/debugging. This is also a feature gap from v2.

#### What is already there? What do you see now?
Last activity values for connected gateways

#### What is missing? What do you want to see?
Instead of showing `DIsconnected` as the gateway status in the gateway overview, we should show `Disconnected since …` in case there has been activity recorded before.

#### How do you propose to implement this?
From what I can see, the Gateway registry itself does not seem to store timestamps of any gateway activity. If that's true, we might have to look through historical events. Any other ideas @adriansmares ?

#### How do you propose to test this?
Based on the implementation, we should be able to mock some API responses to check this in the end-to-end tests.

#### Can you do this yourself and submit a Pull Request?
Passing this one on to @ryaplots. I'm applying the `discuss` label until we determined how to implement this. Please do not start with this yet.

adriansmares, 2021-11-09:
I think it would be 'cleaner' to make the connection stats also persist after the gateway has been disconnected, perhaps for a period such as 24-48 hours. We can then add a field `disconnected_at`, that would be mutually exclusive with `connected_at`, representing the time at which the gateway did disconnect. The frontend can then decide if the gateway is connected based on the field which is present, and if the end point returns 404, it means that the gateway has been inactive for more than two days.

kschiffer, 2021-11-17:
This sounds very good. Would it be a lot of effort to implement that?

kschiffer, 2021-12-06:
Unassigning me for now, since following the last comments, this is more of a IS/GS thing.

NicolasMrad, 2021-12-07:
@adriansmares

adriansmares, 2021-12-07:
The backend implementation is going to be fully GS driven, and requires the following changes:

1. Add a `disconnected_at` field to the `GatewayConnectionStats` message - keep in mind that this is going to be field 11, do not renumber the other ones.
https://github.com/TheThingsNetwork/lorawan-stack/blob/f4058f858fa03242e079b5c884a46e09ca4cb2b5/api/gateway.proto#L365-L394

2. Add a `ttl` parameter to the `Set` call of the gateway connection stats registry 
https://github.com/TheThingsNetwork/lorawan-stack/blob/f4058f858fa03242e079b5c884a46e09ca4cb2b5/pkg/gatewayserver/registry.go#L23-L29
https://github.com/TheThingsNetwork/lorawan-stack/blob/f4058f858fa03242e079b5c884a46e09ca4cb2b5/pkg/gatewayserver/redis/registry.go#L47-L78

You need to propagate the TTL to the underlying Redis command (last parameter is the TTL):
https://github.com/TheThingsNetwork/lorawan-stack/blob/f4058f858fa03242e079b5c884a46e09ca4cb2b5/pkg/gatewayserver/redis/registry.go#L70

3. On gateway disconnect, instead of deleting the entry, we need to just update the `connected_at` to `nil` and `disconnected_at` to `time.Now()`
https://github.com/TheThingsNetwork/lorawan-stack/blob/f4058f858fa03242e079b5c884a46e09ca4cb2b5/pkg/gatewayserver/gatewayserver.go#L884-L889

kschiffer, 2021-12-20:
Can this be closed now?

adriansmares, 2021-12-20:
Yes.