TheThingsNetwork/lorawan-stack #4855: Display gateway connection stats in the Console

adriansmares, 2021-11-09:
#### Summary

It would be nice if the Console would show more gateway connection statistics, if they are available, in the gateway overview form

#### Why do we need this?

Greater gateway debugging capabilities

#### What is already there? What do you see now?

Frame counters (up/down), and last activity meters

#### What is missing? What do you want to see?

- The protocol name
- A small table with round trip times (min/max/median)
- A small table with the sub band utilization quotas

#### Environment

`v3.16`

#### How do you propose to implement this?

Add the fields to the overview page.

#### How do you propose to test this?

Check on `staging1` for the existing gateways how the feature looks.

#### Can you do this yourself and submit a Pull Request?

Can review.

kschiffer, 2023-08-31:
I'll include this in my general update of the Gateway overview.

KrishnaIyer, 2024-10-10:
This is done as part of the redesign

closed 2024-10-10 by KrishnaIyer