TheThingsNetwork/lorawan-stack #4857: Email validation errors with "token already used"

johanstokking, 2021-11-10:
#### Summary

Email validation for invited users errors with an error message that the token was already used.

#### Steps to Reproduce

1. Invite user via CLI
2. User registers
3. User receives email with validation token
4. User clicks link, errors with token already used

Strangely enough:

5. User logins with CLI
6. User copy pastes command from the email
7. Validation succeeds
8. Retry results in the expected error message

This is with the very same token

#### What do you see now?

The user gets an error that the token is already used.

[image]

#### What do you want to see instead?

No error message; the user did not click the link.

Also, when checking the database, it shows that it's not expired and not used.

reference | value | expires_at | used
---|---|---|---
 Y5BSN... | \<snip\> | 2021-11-12 06:01:35.097+00 | f

#### Environment

TTSC `staging1`, so a multi-tenant environment, if that matters

#### How do you propose to implement this?

Not sure, there's something wrong with the link and/or how the account app handles the request

#### How do you propose to test this?

Test on `staging1` with any test account

#### Can you do this yourself and submit a Pull Request?

No

pgalic96, 2021-11-10:
hmmm can't reproduce on the local OS setup, going to try other setups

adriansmares, 2021-11-30:
This is a dual bug between the backend and the frontend.

The frontend provides the parameters to the API client, but the API client does not take parameters, so the request is empty. You can see this with a request that comes via email by watching the networking tab of your browser debugger - the body is empty.

https://github.com/TheThingsNetwork/lorawan-stack/blob/c1affae0d900b5558c9d98249365f946db7d5ab4/sdk/js/src/service/contact-info.js#L22-L26
https://github.com/TheThingsNetwork/lorawan-stack/blob/c1affae0d900b5558c9d98249365f946db7d5ab4/pkg/webui/account/views/validate/index.js#L70-L73

The backend has no validation on the presence of the reference / token, so it will always return the first entity in the tenant:
https://github.com/TheThingsNetwork/lorawan-stack/blob/c1affae0d900b5558c9d98249365f946db7d5ab4/pkg/identityserver/store/contact_info_store.go#L212-L215

pgalic96, 2021-12-01:
Nice catch, will take this from here!

PR #4974 (2021-12-08): Fix account app contact info validation

#### Summary

Closes #4857 

#### Changes

- Add parameters to `validate` call.
- Add `body` to http endpoint for the RPC.
- Add minimum length to proto field requirements to catch empty tokens before they reach the store.
- Enhance e2e tests with a double creation - validation flow.

#### Testing

e2e's should catch all these errors, also manual

##### Regressions

none

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, co [... cut]
closed 2021-12-08 by pgalic96