TheThingsNetwork/lorawan-stack #4871: End device last seen status is not updated on device activity

bafonins, 2021-11-16:
#### Summary

End device last seen status in the Console (end device table) is not updated when sending/receiving traffic. The last seen status is updated only on full page reload

#### Steps to Reproduce

1. Go to the staging environment and select any application with active end devices
2. Observe end devices sending/receiving traffic
3. Observe end device last seen status not updated and displaying the initial value

#### What do you see now?

End device last seen status is not updated live when sending/receiving traffic

#### What do you want to see instead?

End device last seen status should be updated live when sending/receiving traffic

#### Environment

tti staging/latest `v3.16` frontend and tti staging backend

#### How do you propose to implement this?

🤷‍♂️

#### How do you propose to test this?

See reproduce steps

#### Can you do this yourself and submit a Pull Request?

@kschiffer

bafonins, 2021-11-16:
Same behavior is present on the end device overview page (frame counters are not updated as well)

kschiffer, 2021-11-16:
I'll look into it today.

PR #4875 (2021-11-17): Fix resolution of last activity value for end devices and applications
#### Summary
This quickfix PR fixes the "Last activity" value not being updated for end devices and applications.

Closes #4871
Closes #4870 

cc @adriansmares 

#### Changes
- Fix wrong RegExp format for the "heartbeat" filter for end devices.
- Use `as.up.*.forward` events only to determine activity.
- Use `as.up.*.forward` for the verbose filters (instead of listing them one by one)

#### Testing

Mnaual using the staging environment.

#### Notes for Reviewers
The issue was two-fold

1. Since we are using backend filteri [... cut]
closed 2021-11-17 by kschiffer