TheThingsNetwork/lorawan-stack #4874: The end device reset command should also reset the Application Server and Join Server state

adriansmares, 2021-11-16:
#### Summary

The end device `reset` command should also reset the AS and JS states.

#### Why do we need this?

Because right now we reset the NS state, which is not sufficient.

#### What is already there? What do you see now?

Only the Network Server state is reset.

#### What is missing? What do you want to see?

For devices that do not `supports_join`, we should reset `session.last_a_f_cnt_down` in the AS end device, otherwise we may lose a significant chunk of downlinks.
For devices that do `supports_join`, we probably can reset the nonces stored in the JS end device.

#### Environment

`v3.16.0`

#### How do you propose to implement this?

Make sure that the field mask contains `supports_join` while returning the NS end device. This allows the AS/JS logic to take place. Don't forget to remove the field at the end if it was not explicitly requested by the user.

Based on this, update the AS/JS end devices.

#### How do you propose to test this?

Reset an end device.

#### Can you do this yourself and submit a Pull Request?

Can review.

KrishnaIyer, 2022-03-23:
Is this still relevant @adriansmares? If yes, I'll pick it up the next milestone.

adriansmares, 2022-03-23:
Yes, it is still relevant. You can easily test this by adding an ABP device, sending some downlinks, then resetting the MAC state. The frame counter for newer downlinks won't reset.

KrishnaIyer, 2022-06-14:
Moving this to `v3.20.2` when I can have an end device to run tests.

PR #5829 (2022-10-04): Reset Application and Join Server state in CLI reset

#### Summary

Closes #4874

#### Changes

- added `EndDevice.UpdateTimestamps()` method
- `reset` cli command now resets values for `js`/`as` servers as well

#### Testing

Tested locally

- only necessary and requested fields are shown
[image]

##### Regressions

n/a

#### Notes for Reviewers

- [util: Add UpdateTimestamps method to ttnpb.EndDevice](https://github.com/TheThingsNetwork/lorawan-stack/pull/5829/commits/8e6427586eacf1ece14cd306b34e8830b17d3131) makes the `EndDevice.UpdateTimestamps()` method standardised and possibly common wi [... cut]
closed 2022-10-04 by adriansmares