TheThingsNetwork/lorawan-stack #4895: Rename Application / Move device to another application

jpmeijers, 2021-11-19:
#### Summary

* Sometimes one would need to rename the application (mistakes, privacy, permissions, etc). 
* One might also need to move a device from one application to another.

Doing this will allow one to create a new application with the correct name and move the device there, in essence answering the first part.

#### Why do we need this?

Move device to another application without requiring a rejoin.

#### What is already there? What do you see now?

Devices can be claimed. This is a premium feature, and focuses on transferring devices between users. 

#### What is missing? What do you want to see?

A user can't move a device from one application to another.

#### Environment

TTS 3.16.0

#### How do you propose to implement this?

Building on the idea of the claim process, allow one to do a simple transfer of a device between your own applications.

#### How do you propose to test this?

Transfer a device with an active session and see if it keeps on working, delivering data via the new application.

#### Can you do this yourself and submit a Pull Request?

No

jpmeijers, 2021-11-19:
Trying to register the same device under a new applicatiion results in an error:
> Submit failed
> An end device with JoinEUI REDACTED and DevEUI REDACTED is already registered as REDACTED in application REDACTED

KrishnaIyer, 2021-11-23:
Thanks @jpmeijers. We're tracking this in an internal issue but we don't have a milestone yet.

jpmeijers, 2022-01-05:
For a completely different reason I ran into this problem again today. For admin reasons on our side I want to change the name of the application, containing > 10 devices. This is not currently possible.

Last time I had to only move one device, and an export unsing ttn-lw-ctl and import into the new application worked. This time around it will be a lot of effort exporting all the devices, fix the json and import again. So the application has to remain with an incorrect name.

So +1 for this feature.

closed 2025-04-30 by KrishnaIyer