TheThingsNetwork/lorawan-stack #4902: Make CLI retry requests when they failed due to rate limitations

kschiffer, 2021-11-23:
#### Summary
The CLI currently errors out on rate-limiting errors when it should rather wait for the cooldown and retry the request.

#### Why do we need this?
To allow for batch processes under the limitations of rate limiting rules.

#### What is already there? What do you see now?
Various CLI commands that call RPCs

#### What is missing? What do you want to see?
The CLI reacting appropriately to rate limits.

#### How do you propose to implement this?
In the frontend we already apply a middleware that checks for rate-limiting errors and based on the provided information in the response header, waits for a retry of the request.

#### How do you propose to test this?
Manually (?)

#### Can you do this yourself and submit a Pull Request?
I think people more invested into the stack CLI are better suited for this

KrishnaIyer, 2021-11-23:
I can take this one in the next milestone.

nicholaspcr, 2021-12-27:
Decided to create a PR before bringing some questions here, that way I could sort them in a order that makes sense.

I opted to use the [go-grpc-middleware](https://github.com/grpc-ecosystem/go-grpc-middleware) to implement the interceptor and there are two things that might be good to confirm here. 
- The retry interceptor is only used in the `cli` client, I considered adding the retry to the default client but since the retry triggers on the status codes `ResourceExhausted` and `Unavailable` I'm not sure of the impact outside of the cli's scope.
- Currently I didn't add the capability of specifying in the CLI config which codes trigger the retry. I'm in doubt if this is something  that should be customizable or not.

And on another note, I would appreciate it if @KrishnaIyer could give the PR a look. 
Since you were assigned to this previously, I would appreciate hearing how you would tackle the issue.

nicholaspcr, 2022-01-06:
@KrishnaIyer, could you help me out with a behaviour that I found while testing? 

Using the unary interceptor I've been logging the values of the rate limit headers, in specific the headers below:
```
X-Rate-Limit-Limit: The maximum number of allowed requests for the current time period.
X-Rate-Limit-Available: Number of available requests for the current time period.
X-Rate-Limit-Reset: Seconds until the rate limiter resets.
X-Rate-Limit-Retry: Seconds the client should wait before retrying the request.
```
But while testing the values of the  `X-Rate-Limit-Reset` seem to increase overtime instead of decrease, which leads to some confusion in regards to the description in the documentation site. 

I did some tests locally by setting a limit of three requests per minute, then I experimented by:
 - reaching the request limit as soon as possible
 - spacing out the requests around the minute
 - reaching the request limit before the end of the minute
 - waiting 15 seconds and then hitting the request limit

And on all cases after reaching the bump limit the header value seem to always be around the same range, between 50~60 seconds, so I'm wondering if I'm analysing the value in a incorrect manner. By the documentation I was expecting something like a decreasing value that would allow me to known in many seconds the rate limit would reset.

Might help to have the logs of one of the crude tests: https://gist.github.com/nicholaspcr/699aba2a7e5bfa80061d1617abf647ce. This was the last test, in which I waited 15 seconds after the server was initialized before making the requests.

PR #5121 (2022-01-26): Make CLI retry when it fails due to rate limitations

#### Summary

Closes #4902 

#### Changes

- Add unary interceptor for retry request
- Add unary interceptor test cases
- Add interceptor to CLI, disabled by default

#### Testing

Unit testing and local tests

Config and prints:
- [Print of concurrent unary requests with jitter](https://user-images.githubusercontent.com/28912302/150571680-d933fa3c-9b6f-4afb-ae1e-53d9dd815d8b.png)
- cli:
```yml
retry-config:
  max: 5
  jitter: 0.2
  enable_metadata: true
```
- stack
```yml
tls:
  source: file
  root-ca: ./ca.pem
  certificate: ./cert.pem
   [... cut]
closed 2022-01-26 by nicholaspcr