TheThingsNetwork/lorawan-stack #4910: Make Last Activity clickable

jpmeijers, 2021-11-24:
#### Summary

On a device page the Last Activity shows a human readable string like "2 hours ago". On V2 it was possible to click on this string to show the actual timestamp when it was last heard.

#### Why do we need this?

The accuracy of the string is not good enough.

#### What is already there? What do you see now?

[image]

`Last activity 2 hours ago` - Does this mean 1h59m ago, or 1h01m ago?

#### What is missing? What do you want to see?

Like on the V2 console, make this string clickable to change it into displaying the actual timestamp. Maybe as an ISO time string.

#### Environment

V3.16.0

#### How do you propose to implement this?

Use the same component as in the V2 console.

#### How do you propose to test this?

Open a device in the console and click on the last activity time

#### Can you do this yourself and submit a Pull Request?

No

htdvisser, 2021-11-30:
The actual timestamp does appear when you hover over the "xx ago":

[image]

But @kschiffer is going to think about how we can improve this.

jpmeijers, 2021-11-30:
Thanks @htdvisser, somehow I missed that tooltip. It should actually be good enough in most situations. It's only on mobile where this tooltip doesn't work.

kschiffer, 2021-12-06:
To improve the behavior here, the actual date should also be part of the info tooltip. We should simply append to the message

> The last activity was received at Dec 6, 2021 11:27:42.

Can you take care of this @ryaplots ?

PR #5031 (2022-01-03): Append last seen to tooltip info

#### Summary

This pull request closes #4910 by adding the last seen date and time to the tooltip message

#### Changes

- Add to tooltip
- Remove title

[image]

#### Testing

Manually

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: Relevant documentation is added or updated.
- [x] Changelog: Sign [... cut]
closed 2022-01-03 by ryaplots