TheThingsNetwork/lorawan-stack #4937: Let NS/AS update Last Seen of End Device in IS

htdvisser, 2021-11-30:
#### Summary

It would be nice if the Network Server or Application Server could update the "LastSeen" of end devices in the Identity Server. 

#### Why do we need this?

Mostly because users want to sort their end devices on LastSeen, but it's also nice to have for network statistics.

#### What is already there? What do you see now?

LastSeen is tracked in the session of the Network Server.

#### What is missing? What do you want to see?

Either the Network Server or the Application Server should update the LastSeen of the end device in the Identity Server

#### How do you propose to implement this?

- We can add a `last_seen` to the EndDevice API and DB model.
- Since the Application Server already regularly fetches the end device from the Identity Server, I think it's easiest if the Application Server updated the LastSeen in the Identity Server using Cluster Auth. 
  - Updating should happen asynchronously (so not in hot-path)
  - Granularity should be limited; we shouldn't update on every uplink, perhaps only if the LastSeen is more than X hours ago.

adriansmares, 2021-12-02:
This one is clear as well :+1:.

pgalic96, 2022-02-09:
> LastSeen is tracked in the session of the Network Server.

@adriansmares Is this true ? Seems that the session only records the start time. Shouldn't we look at the most recent uplink to determine last seen?

PR #5290 (2022-04-13): Add Device LastSeen

#### Summary

Closes #4937 

#### Changes

- Add `LastSeen` field to `EndDevice` proto and db model
- Add `BatchUpdateEndDevice` to update end devices in batch
- Add `LastSeenProvider` that stores last seen timestamps and updates them in IS periodically and upon the max capacity of the map

#### Testing

- Unit tests

##### Regressions

- New functionality, shouldn't be any.

#### Notes for Reviewers

Does this need to end up in 3.19? There is a db and api change but its all additions.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated  [... cut]