TheThingsNetwork/lorawan-stack #4938: Let NS/AS/JS update DevAddr of EndDevice in IS

htdvisser, 2021-11-30:
#### Summary

It would be nice if the Network Server, Application Server or Join Server could update the DevAddr of end devices in the Identity Server.

Related to #4937.

#### Why do we need this?

Mostly because users want to sort/search their end devices on DevAddr, but it's also nice to have for network statistics.

#### What is already there? What do you see now?

The Network Server, Application Server and Join Server are aware of the DevAddr.

#### What is missing? What do you want to see?

On successful Join, one of these servers should update the DevAddr of the end device in the Identity Server.

#### How do you propose to implement this?

- We should add a `dev_addr` to the EndDevice model in the Identity Server
- The EndDevice API already has `DevAddr` in the `EndDeviceIdentifiers`
- The NS/AS/JS should use Cluster Auth to update the DevAddr in the IS on successful join.

adriansmares, 2021-12-02:
I've just checked the contents of the issue and I don't think that there are missing details here.There are only two clarifications I would add:

- For ABP/Multicast devices, it's probably better if the Console and CLI also fill these fields.
- For ABP we may also provide this information to the Identity Server once we mark the end device as activated. This won't affect all of the existing end device but maybe we can come up with an AS migration for this.

htdvisser, 2022-05-10:
@pgalic96 isn't going to work on this, so re-assigning and bumping this back to triage.