TheThingsNetwork/lorawan-stack #4963: Error: RP002_V1_0_1 does not exist for EU868

itziardelatorre, 2021-12-02:
#### Summary

It seems that RP002-1.0.1 is not supported for LoRaWAN v1.0.4. Note that the LoRaWAN certification v1.0.4 is based on this version of the Regional Parameter document.
According to the example files for the "device repository", this specification should be  the correct one for LoRaWAN v1.0.4. Please refer to "windsensor-profile.yaml":
# LoRaWAN Regional Parameters version. Values depend on the LoRaWAN version:
#   1.0:   TS001-1.0
#   1.0.1: TS001-1.0.1
#   1.0.2: RP001-1.0.2 or RP001-1.0.2-RevB
#   1.0.3: RP001-1.0.3-RevA
#   1.0.4: RP002-1.0.0 or RP002-1.0.1
#   1.1:   RP001-1.1-RevA or RP001-1.1-RevB

#### Steps to Reproduce

Try to create a new device from the device repository configured with those versions: e.g. IMST GmbH, Wireless M-Bus Range Extender, FW V1.1

#### What is already there? What do you see now?

[image]

#### What is missing? What do you want to see?

...

#### Environment

...

#### How do you propose to implement this?

RP002-1.0.1 should be supported for LoRaWAN v1.0.4-

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2021-12-02:
@adriansmares I'm guessing this comes from NS, right?

This might be blocked by #4885

johanstokking, 2021-12-02:
Also references https://github.com/TheThingsNetwork/lorawan-stack/issues/3513

adriansmares, 2021-12-02:
> @adriansmares I'm guessing this comes from NS, right?

Indeed. Will work once #3513 1.0.1 support is done.

closed 2021-12-15 by adriansmares