TheThingsNetwork/lorawan-stack #4971: Maximum downlink payload size exceeded when dwell time is activated

johanstokking, 2021-12-06:
#### Summary

Maximum downlink payload size is exceeded when the end device has dwell time enabled, but the Network Server's frequency plan does not.

#### Steps to Reproduce

1. Register device with `AS_920_923_TTN_AU` (dwell time is deactivated)
2. Configure device with AS923 band, which boots with dwell time activated
3. Let end device join; this works successfully
4. Observe first downlink sent by Network Server

#### What do you see now?

That downlink message can be too long for the end device, which assumes that dwell time is activated. Consequently, the end device may reject the downlink, leading to an invalid state and potential loss of the downlink path.

For instance, it may carry FOpts `06 07 07 a0 af 8c 50 03 31 ff 00 01 09 05`, i.e. `TxParamSetupReq`, `DevStatusReq`, `LinkADRReq` and `NewChannelReq`. This `TxParamSetupReq` deactivates dwell time, but this downlink frame is discarded.

#### What do you want to see instead?

Network Server should not exceed the maximum payload size for downlink, even when dwell time is deactivated in the frequency plan, because it may still be activated in the end device.

#### Environment

TTS 3.16.1, AS923 band, AS_920_923_TTN_AU frequency plan, LoRaWAN 1.0.2, RP1 1.0.2 rev. B.

#### How do you propose to implement this?

NS always considers the dwell time setting from the frequency plan: https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.16/pkg/networkserver/downlink.go#L1243

I think we should instead configure a (default) setting on the band. On MAC state reset, use the band's value for the end device's dwell time in current parameters, and the frequency plan's value in desired parameters. The latter seems already implemented; `TxParamSetupReq` takes the value from the desired parameters, but the problem seems to me that the dwell time used for payload length calculation is always the frequency plan's.

#### How do you propose to test this?

Unit testing

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2021-12-07:
I updated the original comment; we should assume that dwell time is activated in this band and version, but it should be possible to disable it explicitly via a new field in MAC settings. This is because the boot value is undefined in this band and version (which is fortunately fixed in RP2).

johanstokking, 2021-12-07:
@NicolasMrad let's triage this based on @adriansmares assessment.

NicolasMrad, 2021-12-10:
https://github.com/TheThingsNetwork/lorawan-stack/issues/725

johanstokking, 2021-12-13:
> #725

That is since 1.0.3. The behavior in this issue applies specifically to 1.0.2 Rev. B, where the end device's boot parameters in AS923 aren't defined.

#725 is an optimization; avoid sending a redundant `TxParamSetupReq` to deactivate downlink dwell time while it was already deactivated.

This issue and #725 can probably be resolved in one go, once the MAC state reset configure the current parameters correctly. In case of 1.0.2 Rev. B, downlink dwell time should be _activated_ (unless overridden via MAC settings), while in 1.0.3 and higher it should be _deactivated_.

htdvisser, 2022-01-05:
Doesn't look like this is going to get finished for v3.17.0, so pushing to v3.17.1.

closed 2022-02-04 by adriansmares