TheThingsNetwork/lorawan-stack #4972: Contact info validation fails in certain situations

kschiffer, 2021-12-06:
#### Summary

Contact info validation currently fails when multiple validations are currently pending. 

#### Steps to Reproduce

1. Register a user (A) while using the `--is.user-registration.contact-info-validation.required=true` setting.
2. Register another user (B)
3. Wait for the validation email for user (B) and click on the validation link
4. Get a validation failure

#### What do you see now?

Validation failing.

#### What do you want to see instead?

Validation being successful.

#### How do you propose to implement this?

As discovered by @adriansmares, the [`ContactInfo.validate()` function does not take any arguments](https://github.com/TheThingsNetwork/lorawan-stack/blob/c1affae0d900b5558c9d98249365f946db7d5ab4/sdk/js/src/service/contact-info.js#L22), meaning that [the arguments sent to the SDK](https://github.com/TheThingsNetwork/lorawan-stack/blob/c1affae0d900b5558c9d98249365f946db7d5ab4/pkg/webui/account/views/validate/index.js#L70) are discarded. Due to a bug in the IS, the IS will still validate the first contact info validation if the request body is unset, which is also why CI was not failing on when testing contact info validation.

To resolve this issue, we need to pass on the contact info validation body to the API within the SDK.

#### How do you propose to test this?

We should also update the end-to-end test to make sure contact info validation also succeeds when multiple validations are pending.

#### Can you do this yourself and submit a Pull Request?

Yes.

adriansmares, 2021-12-06:
Duplicate of https://github.com/TheThingsNetwork/lorawan-stack/issues/4857 ?

kschiffer, 2021-12-06:
Indeed, will merge this.

closed 2021-12-06 by kschiffer