TheThingsNetwork/lorawan-stack #4982: Automate two step release process

adriansmares, 2021-12-10:
#### Summary

Our current release process has two steps:
- Create and tag the actual release
- Goreleaser actually generates the release artifacts (archives, binaries, docker images and so forth) based on the tag

We should also automate it in order to have an easier time releasing features.

#### Why do we need this?

The first step is manual and documented in [RELEASE.md](https://github.com/TheThingsNetwork/lorawan-stack/blob/v3.16/.github/ISSUE_TEMPLATE/release.md), and creates a small barrier to releasing versions. I believe that if everything is automated via one or more GitHub actions, and the person responsible with the release only has to provide the commit hash and the bump type, then review the PR, things would go a lot faster and we could actually have release candidates before releasing versions that we distribute.

#### What is already there? What do you see now?

The manual steps of `RELEASE.md`, and @htdvisser's automation work in this commit https://github.com/TheThingsNetwork/lorawan-stack/commit/86141a66a6410cc581e6925b43d0be4a6da353e3

#### What is missing? What do you want to see?

I want the manual step to be automated - I just want to provide a hash and the bump type.

#### Environment

N/A

#### How do you propose to implement this?

A GitHub action should be able to automate this.

#### How do you propose to test this?

Do some `bumpMajor` releases, we can delete those anyway.

#### Can you do this yourself and submit a Pull Request?

I can review.

michalborkowski96, 2022-01-18:
I was thinking about it... now the release looks like this:

1. Run a github action
2. Merge a PR
3. Run another github action
4. Merge another PR
5. Run yet another github action

So, it's rather automatic at this point, save a few PRs here and there. The thing we wanted to fix is another problem:

1. We release on Friday
2. We deploy on Monday morning
3. We patch on Monday afternoon
4. We inform customers that version 3.x.y has a bug and shouldn't be used

So, we want to kinda reverse the order: we want to deploy and then release. Of course, in order to deploy we need to have a release, which creates a circular dependency. The solution would be to split the release into two steps, and make it work like this:
1. Internal release
2. Deploy
3. Publish images

In case of a bug, we could simply repeat steps 1 and 2:
1. Internal release 3.x.y
2. Deploy
3. Discover a bug
4. Patch it, overwrite docker tag 3.x.y in our private docker repo
5. Deploy again
6. Make 3.x.y public

How do we achieve this?

Looking from the perspective of "what needs to be done", release looks like this:
1. Update a bunch of files that contain the version number.
2. `git submodule update --init --remote`
3. Update `CHANGELOG.md`
4. Build and push images
5. Update version in `lorawan-stack-aws`
6. Build and push AWS templates

First, I'd like to point out that `git submodule update --init --remote` doesn't have to be run on release. We don't care whether they contain the latest commits or not - all we want is that submodules are reasonably recent, whatever this means. Therefore, instead of having this as a part of the release process, we can have a github action that runs weekly and commits `git submodule update --init --remote`.

Second, we don't need version number in the repo. We need to know the version number when building the image, but we don't actually need it to be checked out in the repo. We can remove this file from the repo, and instead generate it right before build, based on github action argument. Someone clicks "I want an image with this commit as version 3.x.y" and github action creates an image 3.x.y. There is one problem: currently snapshots have versioning like `3.x.y-SNAPSHOT-hash`. This would break (CI wouldn't be aware that internal release has already been done). We would need to have a discussion how to handle that, but since snapshots aren't something that customers are supposed to actually use, I don't see this as a problem. There are multiple solutions: changing version scheme, skipping the patch version number in snapshots, adding github secret with most recent internal release, or simply accepting the fact that snapshots build between internal release and public release would have slightly wrong version.

Third, we don't need to update `CHANGELOG.md` at the moment of image build. We can do it retroactively later.

Therefore, possible workflow would be like this:
1. (Friday) Build and push images to `lorawan-stack-rc`
2. (Monday-Tuesday) Deploy, if there are bugs repeat step 1, potentially overwriting said images
3. (Wednesday) Mirror images from `lorawan-stack-rc` in `lorawan-stack`, update `CHANGELOG.md`, publish information about new version

My plan doesn't consider `lorawan-stack-aws`, but since the pace of changes there is much slower (I can manually stall release), we can ignore that now and get back to it later.

michalborkowski96, 2022-01-18:
@adriansmares @htdvisser

michalborkowski96, 2022-01-20:
Discussion moved to: https://github.com/TheThingsNetwork/lorawan-stack/issues/4982

michalborkowski96, 2022-02-16:
I move this to next milestone. The PR has been merged, but let's keep this in mind for a moment so that we can see if there's something to be improved.

closed 2022-03-07 by michalborkowski96