TheThingsNetwork/lorawan-stack #5027: Enable selecting users via autosuggest

kschiffer, 2021-12-20:
#### Summary
We currently only allow specifying users via regular text input. This is inconvenient because it forces users to remember and/or copy user ids from other places. Instead the user/org input should autosuggest available users when typing into it.

#### Why do we need this?
Improve UX of adding collaborators, contacts, owners, etc.

#### What is already there? What do you see now?
- User/org input via simple text input
- Search RPC for searching users/orgs `SearchOrganizations` and `SearchUsers`

#### What is missing? What do you want to see?
We should introduce a `<UserInput />` component specifically for selecting users and/or orgs. It should allow specifying whether it is possible to select only users or both users and organizations.

#### How do you propose to implement this?
I think we can introduce a utility factory/HOC for this which allows us to generate specific auto-suggested inputs for various use cases. This would be similar to the `<FetchSelect />` container we currently use. It should:
1. Take a handler that takes in the current (debounced) value and return a list of suggestions (typically by piping it through a search API request)
2. Pipes the values from 1. and passes them as options to a `<Select />`

We can then use such component generator to define a `<UserInput />` that will autosuggest users and orgs. Similar to how we use `<FetchSelect />` with e.g. `<FreqPlansSelect />`.

#### How do you propose to test this?
End-to-end tests.

#### Can you do this yourself and submit a Pull Request?
@ryaplots

htdvisser, 2021-12-21:
The existing user search RPC is restricted to admins only, so this won't work for regular users.

We should probably think about a separate "account ID autocomplete / autosuggest" API that understands the constraints (because for #4996, entity contacts can only be accounts that are collaborators on the entity).

kschiffer, 2023-04-26:
This was closed by accident.

kschiffer, 2023-04-26:
Here's the UX for this component:
[image]

Some remarks for the implementation:

`ReactSelect` should be able to support this. For choosing a user/org in scoped mode, we'd need first fetch the first 8 accounts of the selected scope (using the regular List RPCs) and then switch to the AccountSearch API as soon as the first two characters are typed in.

@ryaplots please implement this component as a prerequisite for #6112. We'll then also change the old collaborator forms to use this new component.

PR #6306 (2023-07-07): Enable selecting users via autosuggest

#### Summary

Closes #5027 

#### Changes

- Add `<UserInput />` component
- Add `<AutoSuggest />` component
- Add search store and sdk
- Refactor Collaborator form to use `<UserInput />`

<details>
<summary>Screenshots</summary>
- Regular autosuggest
[image]
[image]

- Collaborator limited
[image]

- Multi select
[image]
</details>

#### Testing

cypress

#### Checklist

- [x] S
closed 2023-07-07 by ryaplots