TheThingsNetwork/lorawan-stack #5045: EUIStore.CreateEUIBlock Updates Existing EUI Block

htdvisser, 2021-12-27:
#### Summary

Unlike the name suggests, the `CreateEUIBlock` method of the current `EUIStore` implementation may change an existing EUI block.

#### What do you see now?

https://github.com/TheThingsNetwork/lorawan-stack/blob/dfa369c4457a74b0c04684684a6b8f522e83fb4d/pkg/identityserver/store/eui_block_store.go#L115-L140

#### What do you want to see instead?

A method that suggests we're creating something, shouldn't update something.

#### How do you propose to implement this?

Re-implement.

#### Can you do this yourself and submit a Pull Request?

Sure

pgalic96, 2022-02-08:
@htdvisser What do you exactly expect from this functionality?

The initial idea was to have a function that checks if the EUI block of the right type exists in the database, if it doesn't then it creates a new one (based on config), and if it does it checks whether the db block is the same as the one in config. If it is, we do nothing, and if it isn't, we update the block to match the config.

I would say this is just a naming issue. `InitializeEUIBlock` might be a better name. Then we can move these three functionalities (Get, Create, Update) to unexported functions which can then be exported (and tested) by whoever needs them at that time. Right now we don't need these methods specifically.

htdvisser, 2022-02-09:
Why wouldn't we make the EUI Store store multiple blocks? That way, when we (almost) exhausted all EUIs in the current block, we can add a new block and continue issuing EUIs without interruption.

pgalic96, 2022-02-09:
That was the idea with changing the block in the config, but I agree, overwriting the existing block instead of just creating a new one is quite a messy procedure and it doesn't allow us to possibly reuse unused DevEUI's at a later stage.

I suppose this would mean we should pass a list of prefixes in the config. Upon IS startup we create all blocks (so we don't need to worry about creating them later upon reaching the limit of the current block). When issuing DevEUI for an application we check for an available EUI in all blocks of the same type (`dev_eui`). We error if all blocks are full.

PR #5199 (2022-02-15): Allow for multiple block creation for the same eui block type

#### Summary

Closes #5045 

#### Changes

- Remove unique constraint and index for the `eui_type` field
- Updating config creates a new block, not overwriting the old one
- Issuing a dev block now selects all blocks that did not reach the limit, and picks a first one to issue DevEUI from

#### Testing

unit

##### Regressions

should be covered by tests

#### Notes for Reviewers

This will require a migration because the `type` was unique

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated chang [... cut]
closed 2022-02-15 by pgalic96