TheThingsNetwork/lorawan-stack #5051: downlink_count in GatewayConnectionStats is missing

sanijs, 2021-12-28:
#### Summary

When using the Gateway Server API only uplink_count is available, downlink_count is no longer there 

#### Steps to Reproduce

Use the following API
/api/v3/gs/gateways/{gateway_id}/connection/stats

#### What do you see now?

Everything expect downlink_count

#### What do you want to see instead?

downlink_count

#### Environment

The Things Stack Cloud v3.16.2

#### How do you propose to implement this?

Find the bug and fix it :)

#### How do you propose to test this?

Use the API and see if you get downlink_count

#### Can you do this yourself and submit a Pull Request?

no

sanijs, 2021-12-29:
Upon closer inspection I found out that this only happens when downlink count is 0, if it's more than 0 it works like it should. Is this still a bug or is it meant to work that way?

downlink_count > 0
```
{
   "connected_at":"2021-12-28T16:52:18.033055340Z",
   "protocol":"ws",
   "last_status_received_at":"2021-12-28T16:52:18.061257898Z",
   "last_status":{
      "time":"2021-12-28T16:52:18.061116183Z",
      "versions":{
         "platform":"browan_mt7620a - Firmware 1.0.0 - Protocol 2",
         "station":"2.0.5(browan_mt7620a/std)",
         "firmware":"1.0.0",
         "package":"1.0.0"
      },
      "advanced":{
         "features":"rmtsh",
         "model":"browan_mt7620a"
      }
   },
   "last_uplink_received_at":"2021-12-29T08:10:07.035543275Z",
   "uplink_count":"330",
   "last_downlink_received_at":"2021-12-29T08:09:36.015510341Z",
   "downlink_count":"4",
   "round_trip_times":{
      "min":"0.030961094s",
      "max":"0.032887080s",
      "median":"0.031208629s",
      "count":8
   },
   "sub_bands":[
      {
         "min_frequency":"863000000",
         "max_frequency":"865000000",
         "downlink_utilization_limit":0.001
      },
      {
         "min_frequency":"865000000",
         "max_frequency":"868000000",
         "downlink_utilization_limit":0.01
      },
      {
         "min_frequency":"868000000",
         "max_frequency":"868600000",
         "downlink_utilization_limit":0.01,
         "downlink_utilization":0.00054357335
      },
      {
         "min_frequency":"868700000",
         "max_frequency":"869200000",
         "downlink_utilization_limit":0.001
      },
      {
         "min_frequency":"869400000",
         "max_frequency":"869650000",
         "downlink_utilization_limit":0.1
      },
      {
         "min_frequency":"869700000",
         "max_frequency":"870000000",
         "downlink_utilization_limit":0.01
      }
   ]
}
```
downlink count = 0
```
{
   "connected_at":"2021-12-28T10:15:10.524016520Z",
   "protocol":"ws",
   "last_status_received_at":"2021-12-28T10:15:10.569662222Z",
   "last_status":{
      "time":"2021-12-28T10:15:10.569518066Z",
      "versions":{
         "station":"2.0.5(browan_mt7620a/std)",
         "firmware":"1.0.0",
         "package":"1.0.0",
         "platform":"browan_mt7620a - Firmware 1.0.0 - Protocol 2"
      },
      "advanced":{
         "features":"rmtsh",
         "model":"browan_mt7620a"
      }
   },
   "last_uplink_received_at":"2021-12-29T06:29:47.306236672Z",
   "uplink_count":"30",
   "round_trip_times":{
      "min":"0.045700217s",
      "max":"0.046062636s",
      "median":"0.046032458s",
      "count":3
   }
}
```

johanstokking, 2021-12-29:
@sanijs I was meant to comment that here indeed. Currently we use protobuf JSON marshaling where empty fields are omitted. We are in a long process of addressing this, along with many other things.

References #2258
References #2298
References #2798

johanstokking, 2021-12-29:
Blocked on #2798 and a not-yet-filed issue to include field masks in responses so our JSON marshaler knows which (empty) fields to return.