TheThingsNetwork/lorawan-stack #5062: Update to version 3.16.2 breaks webhooks with header values longer then 256 characters

ff6347, 2022-01-03:
#### Summary
Since a few days before Christmas we are getting the following error in the console when creating webhooks.

```json
{
  "code": 3,
  "message": "error:pkg/errors:validation (invalid `webhook`: embedded message failed validation)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/errors",
      "name": "validation",
      "message_format": "invalid `{field}`: {reason}",
      "attributes": {
        "field": "webhook",
        "name": "SetApplicationWebhookRequestValidationError",
        "reason": "embedded message failed validation"
      },
      "correlation_id": "5523f016da024c79a8def290bd0e5d44",
      "cause": {
        "namespace": "pkg/errors",
        "name": "validation",
        "message_format": "invalid `{field}`: {reason}",
        "attributes": {
          "field": "headers[Authorization]",
          "name": "ApplicationWebhookValidationError",
          "reason": "value length must be at most 256 runes"
        },
        "correlation_id": "0676cd6a06f94680af3e3fad394befdd",
        "code": 3
      },
      "code": 3
    }
  ]
}
```

We are using JWTs to authenticate and enable users to send data from TTN to our platform. These JWTs are longer than 256 characters. I think I found the commit that introduced this validation/resctriction. https://github.com/TheThingsNetwork/lorawan-stack/commit/f57027aefdb2a474610ac6dbb5cd4cd413a8a5fd

And below is the link to the API docs.

https://github.com/TheThingsNetwork/lorawan-stack/blob/f57027aefdb2a474610ac6dbb5cd4cd413a8a5fd/api/api.md#field-rules-44

Is there a reason for this limitation?

#### Steps to Reproduce

1. Create an application
2. Create a device (use advanced Activation by personalization (ABP))
3. Go To Integrations/Webhooks/custom and create a new one
4. Add Base URL
5. Add a Header with more then 256 characters e.g. `Authorization: Bearer qwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewqqwpoiuztrewq1234`

#### What do you see now?

[image]

#### What do you want to see instead?

That the webhook was created

#### How do you propose to implement this?

Increase or remove the validation max length in the file api/applicationserver_web.proto (if this is the right place for that)

#### How do you propose to test this?

Unit tests should create random header values longer then 256 characters

#### Can you do this yourself and submit a Pull Request?

I'm not used to golang but I could give it a try.

KrishnaIyer, 2022-01-04:
Thanks @ff6347. We don't really see why an HTTP auth token would need to be more than 256 characters long but we also don't find a need to restrict it to that number. In the next milestone, we'll see if we can bump this to 4k.

ff6347, 2022-01-04:
Hi @KrishnaIyer
Thanks for considering this. We really appreciate it.

If the tokens are opaque I agree. There is no need for such a long string. We are using JSON Web Tokens. For example I created a token with just the optional registered claims as defined in [RFC 7519 section 4.1](https://datatracker.ietf.org/doc/html/rfc7519#section-4.1) and I already get 315 characters

```text
eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJzdWIiOiIxMjM0NTY3ODkwIiwibmFtZSI6IkpvaG4gRG9lIiwiaWF0IjoxNTE2MjM5MDIyLCJpc3MiOiJ0ZWNobm9sb2dpZXN0aWZ0dW5nIiwiYXVkIjoic3VkbyIsImV4cCI6MTUxNjIzOTAyMywibmJmIjoxNTE2MjM5MDIxLCJqdGkiOiI0Nzc0Y2IyMC0yNjdkLTRjZDAtOGI1MS04ZTJiNmI0MzFhYzEifQ.Go7FdDSSMf4HwjV46Alcb5Df3QanjWBMF5kfWc3cR4E
```

Decoded payload:

```json
{
  "sub": "1234567890",
  "name": "John Doe",
  "iat": 1516239022,
  "iss": "technologiestiftung",
  "aud": "sudo",
  "exp": 1516239023,
  "nbf": 1516239021,
  "jti": "4774cb20-267d-4cd0-8b51-8e2b6b431ac1"
}
```

You can decode it yourself here at https://jwt.io/

A token issued by e.g. supabase.com already has around 350 characters and the payload is not that complex either.

closed 2022-01-17 by KrishnaIyer

ff6347, 2022-01-17:
Thank you. We really appreciate it.

PR #5109 (2022-01-17): Extend Webhook header value length

#### Summary

Closes #5062 

#### Changes

- Extend Webhook header value length to 4096.

#### Testing

- [x] UT 
- [x] Console set test
[image]

##### Regressions

I don't think there's a regression.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [ ] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [x] Documentation: Relevant documentation is added or updated.
- [x [... cut]