TheThingsNetwork/lorawan-stack #5085: Device location in database of NaN, NaN crashes console, CLI won't allow update

descartes, 2022-01-11:
#### Summary
hiveiq-eu-otaa-01 in application hiveiq-eu-testing appears to be set to NaN, NaN, so won't display as leaflet.js falls over. 

And the CLI raises an exception when trying to set using --location.latitude & --location.longitude and reset --locations returns the values back as NaN, NaN. 

Other devices don't return any JSON values for location (presumeably the ProtocolBuffers quirk)

#### Steps to Reproduce
Have a look on the console - I'll try and setup a new device for the testing so you've got something to play with.

#### What do you see now?

```
instrument.js:109 Error: Invalid LatLng object: (NaN, NaN)
    at new I (leaflet-src.js:1365:11)
    at N (leaflet-src.js:1456:13)
    at B.extend (leaflet-src.js:1152:31)
    at new B (leaflet-src.js:1124:10)
    at R (leaflet-src.js:1335:11)
 
etc
etc
```

and 

```
nick@CatMac ~ % ttn-lw-cli end-devices set hiveiq-eu-testing hiveiq-au-otaa-01 --location.latitude 53.0 --location.longitude -2.0
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x210e81e]

goroutine 1 [running]:
go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.setEndDevice(0xc00095ac00, {0xc00098f5a0, 0x1, 0x1091033}, {0xc00098f5b0, 0x104ac72, 0x860}, {0xc00098f5c0, 0x0, 0x1}, ...)
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/end_devices_split.go:204 +0x4fe
go.thethings.network/lorawan-stack/v3/cmd/ttn-lw-cli/commands.glob..func85(0x376eae0, {0xc00009c660, 0x2, 0x6})
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/commands/end_devices.go:705 +0x11bf
github.com/spf13/cobra.(*Command).execute(0x376eae0, {0xc00009c600, 0x6, 0x6})
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:856 +0x60e
github.com/spf13/cobra.(*Command).ExecuteC(0x377ba60)
	/home/runner/go/pkg/mod/github.com/spf13/cobra@v1.2.1/command.go:974 +0x3bc
main.main()
	/home/runner/work/lorawan-stack/lorawan-stack/cmd/ttn-lw-cli/main.go:27 +0x25
```

#### Environment
macOS Catalina, tried up to date versions of Chrome, Safari & FireFox.

nick@CatMac ~ % ttn-lw-cli version 
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.17.0
Build date:          2022-01-07T10:26:01Z
Git commit:          295590743
Go version:          go1.17.5
OS/Arch:             darwin/amd64

kschiffer, 2022-01-17:
Also assigning @htdvisser for the CLI issue.

htdvisser, 2022-01-17:
That CLI error was fixed by @KrishnaIyer in https://github.com/TheThingsNetwork/lorawan-stack/pull/5088.

kschiffer, 2022-01-24:
This can be closed then.

closed 2022-01-24 by kschiffer