TheThingsNetwork/lorawan-stack #5092: Allow overriding the maximum number of retransmissions

adriansmares, 2022-01-12:
#### Summary

The end device MAC settings should allow the user to limit the max `NbTrans` used during ADR.

#### Why do we need this?

The `NbTrans` parameter controls the number of transmissions done for a singular LoRaWAN frame. It is by default one, and is increased when the Network Server detects gaps in the uplink frame counters. It adds a degree of redundancy which is meant to help with the loss rate experienced by the end device. 

This mechanism is abstracted away from the application layer - the application sees the message only one time. But, if the application layer protocol does retransmissions on its own, the `NbTrans` mechanism is itself redundant and can cause significant costs in battery life.

#### What is already there? What do you see now?

A static maximum of 3 transmissions
https://github.com/TheThingsNetwork/lorawan-stack/blob/10da1157cb27216d89e651a43750686bb9ea5ea4/pkg/networkserver/mac/adr.go#L71-L72

#### What is missing? What do you want to see?

1. A MAC setting that allows the user to limit the `NbTrans`. I don't think we should allow the rates themselves to be editable, so this setting should be at most 3.
2. Take into account the MAC setting during the `NbTrans` computation
https://github.com/TheThingsNetwork/lorawan-stack/blob/10da1157cb27216d89e651a43750686bb9ea5ea4/pkg/networkserver/mac/adr.go#L287-L301

#### Environment

`3.17.0`

#### How do you propose to implement this?

Add the API option and adapt the ADR code to take it into account, similar to how other MAC settings are taken into account (separate function that takes into account overrides such as the NS settings).

#### How do you propose to test this?

Unit testing.

#### Can you do this yourself and submit a Pull Request?

Yes. cc @htdvisser / @johanstokking if there are any comments on this.

johanstokking, 2022-01-13:
Wouldn't we want to implement this with https://github.com/TheThingsNetwork/lorawan-stack/issues/40?

I think it becomes tedious to configure this per end device.

adriansmares, 2022-01-13:
But the profiles are on a per-device basis too, right ?  While these kind of things could become part of the profile, you still need to configure this on a per-device basis.

closed 2022-04-12 by adriansmares