TheThingsNetwork/lorawan-stack #5110: Class C Downlink with wrong DevAddr when no uplink was sent before

niondir, 2022-01-18:
#### Summary
When sending a Calss C downlink after a Join, the downlink is scheduled with an invalid (old) DevAddress.

When the device sends an uplink, the issue is gone.

#### Steps to Reproduce

1. Join device that does not send any uplink and switches to class C
2. Join shows DevAddress (e.g. 260B1049)
3. Schedule a Class C Downlink
4. Downlink shows wrong (old) DevAddr (e.g. 260B1C7B)

[image]
[image]

#### What do you see now?
From my Device Log I see:

```
Successfully joined a LoRaWAN network.
Successfully switched to class C
LORAWAN_EVENT_MAC_PENDING
MAC ADDRESS Error: (stack) 260B1049 != 260B1C7B (frame)
```

#### What do you want to see instead?
A valid downlink on my device

#### Environment
LoRaWAN Stack: LoRa Mac Node
Public TTN Network v3.16.2

#### How do you propose to implement this?

#### How do you propose to test this?

#### Can you do this yourself and submit a Pull Request?
No.

adriansmares, 2022-01-21:
This is expected behavior - the Network Server cannot know if the end device has actually received the Join Accept, and as such considers the new session as pending until the end device signals that the Join Accept has been received by confirming the session using an uplink.

closed 2022-01-21 by adriansmares

niondir, 2022-01-23:
Is it reasonable, that a device sends a join request while maintaining the old session? Can't the network server assume that a join request is only issued when the old session is invalid?

At least for the device's I developed that is always the case. But I would need to lookup the spec to argue about that.

adriansmares, 2022-01-24:
> Is it reasonable, that a device sends a join request while maintaining the old session? Can't the network server assume that a join request is only issued when the old session is invalid?

For a 1.0.x device, this is indeed very unlikely to happen. But for example for 1.1.x device, it may issue a _rejoin_ request which indeed may cause the end device to still use the old session. 

I can agree that the behavior is not ideal, and just emptying the queue is not going to help anyone. I will track a separate issue in order to basically delay the class C downlinks until the session is confirmed (i.e. an uplink from the new session arrives), or at least we receive traffic from the old session (for 1.1 situations).