TheThingsNetwork/lorawan-stack #5133: Missing end devices when sorting by name

sanijs, 2022-01-24:
#### Summary

I have a lot of end devices with the same name, when I sort by name some of them will not appear in the list. It changes if I sort by ascending or descending. If I do a search by ID I can find the end device in my list. If I go trough all the pages of end devices and count them, I will get the correct number compared to the number that should be in the application. 

#### Steps to Reproduce

Create a lot of end devices in a application with the same name, or maybe do check in my application?

#### What do you see now?

If I sort by ascending, some of my end devices will not show up when I go through the pages of end devices in my application.

#### What do you want to see instead?

All my end devices when sorting by name

#### Environment

The Things Stack Cloud v3.16.2 (I think it should be higher, but version number is not updated in the console)

#### How do you propose to implement this?

don't know

#### How do you propose to test this?

Sort by name and see if all devices will show up

#### Can you do this yourself and submit a Pull Request?

no

NicolasMrad, 2022-01-25:
Please submit a ticket with this issue using the support portal.

sanijs, 2022-01-28:
Ticket submitted: TTS-4029

PR #5256 (2022-02-24): Fix ordering of entities with same values in the sort field

#### Summary

This pull request adds a second order expression for lists with user-specified order, so that (for example) entities with equal (or without) names are still ordered by their ID.

This fixes #5133.

Also closes https://github.com/TheThingsIndustries/lorawan-stack-support/issues/707.

#### Changes

- Add secondary order expression when not sorting by the default field

#### Testing

🐒 

##### Regressions

none expected

#### Notes for Reviewers

nil

#### Checklist

- [x] Scope: The referenced issue is addressed, there ar [... cut]
closed 2022-02-24 by htdvisser