TheThingsNetwork/lorawan-stack #5150: Improve webhook health status Console UX

adriansmares, 2022-01-26:
#### Summary

References https://github.com/TheThingsNetwork/lorawan-stack/issues/4526
References https://github.com/TheThingsNetwork/lorawan-stack/pull/5002

The Console should show information in a more contextual manner to the user. Specifically, the webhook should not be shown as 'pending' when the health checks are completely disabled.

#### Why do we need this?

Right now, even though the checks are completely disabled, the webhooks are shown as pending, and users are told that they will move from pending to healthy when the webhook will be executed. This is wrong, as no health check will take place if they are disabled.

#### What is already there? What do you see now?

The health status is pending in the webhook list, and editing the webhook tells the user that the status will change soon.

#### What is missing? What do you want to see?

I have described the UX that I had in mind at the bottom of [this](https://github.com/TheThingsNetwork/lorawan-stack/issues/4526#issuecomment-988697068) comment, specifically:

> The frontend may provide quite some contextual help.  For example, if `failed_attempts` is smaller than `unhealthy_attempts_threshold`, the Console may convey that the next HTTP request will be attempted, but if it fails the number of attempts will increase.

> If `failed_attempts` is greater or equal to `unhealthy_attempts_threshold`, then the `last_failed_attempt_at` can be used to relay what will happen to the next request. If `last_failed_attempt_at` + `unhealthy_retry_interval` is a moment in the past, the request will be executed, but if it is a moment in the future, the request will be dropped (no attempt will be made).

> It can also be the case that `unhealthy_attempts_threshold` is `0`. This means that the Application Server will be doing strictly book keeping - the number of attempts will increase, the timestamp will be updated, the last error will be stored, but the requests will always be made - basically we're only monitoring the webhook. In such cases I would recommend that the Console conveys a friendly message that tells the users to get their webhook fixed asap or at least delete it if it's not used.

#### Environment

`3.17.2`

#### How do you propose to implement this?

First of all, backport the AS settings retrieval from enterprise:

https://github.com/TheThingsIndustries/lorawan-stack/blob/0906e8b33239cd4eff2769151fbf22e352a97835/pkg/webui/console/store/middleware/logics/application-server.js

This will allow the UI to have access to the health configuration, and in turn to provide this contextual information.

#### How do you propose to test this?

Edit the stack settings and send some successful/failed webhooks. I've given some advice in my above mentioned comment.

#### Can you do this yourself and submit a Pull Request?

I can review. @kschiffer if the proposed UX looks good, please remove the label.

adriansmares, 2022-02-10:
This will hide the webhook as being possibly unhealthy until the threshold is hit, but that's ok.

kschiffer, 2022-02-10:
We could add another `Recent failures` status when failures were detected but we're still under the threshold. How long are failures retained though? I would want to avoid to have the status being shown as such when there was one failed request one week ago.

Also, do I see correct that the status will jump immediately to `Unhealhy` if the very first request failed?

kschiffer, 2022-02-22:
Closing this via #5222, which resolves the immediate issue. We can look at providing more info to the user after evaluating the current solution.

PR #5222 (2022-02-22): Fix webhook status UX
#### Summary

This PR fixes UX issues with the new webhook health status in the Console.

Closes #5150 

[image]

#### Changes

- Change wording to `Healthy`, `Pending` and `Requests failing`
- Check whether webhook health checks are enabled and do not show any status indication when health status is disabled
- Add AS config service to JS SDK

#### Testing

Manual testing on staging environment.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, stora [... cut]
closed 2022-02-22 by kschiffer