TheThingsNetwork/lorawan-stack #5170: Timeout web console live data layouts to something static to save TTI servers un-watched traffic

descartes, 2022-02-03:
#### Summary
Timeout a console window if it is showing any form of live data to some resting place - or pause the feed

#### Why do we need this?
So Chrome doesn't kill your servers & my machine when I leave too many tabs open.

This is particularly problematic (mainly for TTI) when I've got three or four machines with various tabs running ...

#### What is already there? What do you see now?
I'm trying to get in to the habit of clicking Applications at the top of the window, but it's down to me to remember.

#### What is missing? What do you want to see?
Perhaps an embedded timeout that can be adjusted on the user settings page and on the application page.

I'd be happy for it to be 60 minutes by default and be able to set it up to 12 hours (longer than a normal working day) on a per application basis.

#### How do you propose to implement/test/DIY this?
I think this is an @kschiffer thing

descartes, 2022-10-19:
This strikes me as a win for all parties - I don't toast my workstations with lots of console tabs left open, you don't have the servers toasted serving console data to no one watching.

kschiffer, 2022-10-19:
I agree that some solution is needed here. My initial idea was to automatically pause live data tabs that are not active but that might be a bit too aggressive. Probably, a timeout for unfocused tabs will be good. l don't know if we need any restrictions for focused live-data tabs. If a user wants to actively watch the live data for hours on end, they should be good to do so.

descartes, 2022-10-19:
Unfocused tabs after a period of time are the real issue - as I walk away from my monitors I can see what's running live data which I usually leave open to capture info. But as I have Safari & Chrome on my iMac and Chrome on my iWin (an iMac running Windows 10) and usually have ~10 windows with ~20 tabs on each, losing track of random tabs that are receiving events is oh so easy.

kschiffer, 2022-10-20:
> That said, sometimes 10,000 console lines would be useful!

Yeah it's a thin edge here. I believe once we have proper filtering in place that will also solve many issues here since users can choose to not ingest any irrelevant events.

descartes, 2022-10-20:
I have that as a standalone client side page since Feb '21 - but because it needs an API key, even a read only one, people get in a tangle thinking I'm gaining access to their bank account I don't share it as support is a mess.

KrishnaIyer, 2025-04-30:
We have filtering in place and console events have a dedicate API.

closed 2025-04-30 by KrishnaIyer