TheThingsNetwork/lorawan-stack #5173: Remove gateway location from Packet Broker Mapper not working

mauro742, 2022-02-03:
#### Summary
I am trying to remove the location from a gateway. From the Console, I have unset the "Make location public" setting. However, the location is not is not removed from Packet Broker Mapper. The issue persists also deleting all the location data from the Console. The fix introduced in v3.17.1 (issue #5079) does not solve the problem.

...

#### Steps to Reproduce

1. Set the gateway location in the Console and set the "Make location public" flag
2. After the propagation of the location to Packet Broker Mapper, unsed the "Make location public" flag

#### What do you see now?

...

#### What do you want to see instead?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

...

johanstokking, 2022-02-08:
@mauro742 has the gateway been online since you set the location public off?

mauro742, 2022-02-08:
@johanstokking yes, the gateway was always online. In the following, I report the gateway live data streaming after the public location unsetting:

`[
  {
    "name": "gs.status.receive",
    "time": "2022-02-08T11:12:03.904231147Z",
    "data": {
      "@type": "type.googleapis.com/ttn.lorawan.v3.GatewayStatus",
      "time": "2022-02-08T11:12:03Z",
      "versions": {
        "ttn-lw-gateway-server": "3.17.2"
      },
      "metrics": {
        "ackr": 100,
        "rxfw": 0,
        "rxin": 0,
        "rxok": 0,
        "txin": 0,
        "txok": 0
      }
    },
    "origin": "ip-10-100-12-248.eu-west-1.compute.internal"
  },
  {
    "name": "gs.gateway.connect",
    "time": "2022-02-08T11:12:00.149515779Z",
    "origin": "ip-10-100-12-248.eu-west-1.compute.internal"
  },
  {
    "name": "gs.gateway.disconnect",
    "time": "2022-02-08T11:11:28.258470315Z",
    "data": {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/gatewayserver",
      "name": "gateway_changed",
      "message_format": "gateway changed in registry",
      "correlation_id": "427325ab84404b3c813da007959558a6",
      "code": 2
    },
    "origin": "ip-10-100-12-248.eu-west-1.compute.internal"
  },
  {
    "name": "gateway.update",
    "time": "2022-02-08T11:09:30.148691622Z",
    "data": {
      "@type": "type.googleapis.com/google.protobuf.Value",
      "value": [
        "location_public"
      ]
    },
    "origin": "ip-10-100-6-27.eu-west-1.compute.internal"
  }
]`

PR #5194 (2022-02-08): Reset gateway location in Packet Broker Mapper

#### Summary

Closes #5173 

#### Changes

- Explicitly set empty location to signal that the location needs to be reset

#### Testing

Local integration testing

##### Regressions

None

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: Relevant documentation is added or updated.
- [x]  [... cut]
closed 2022-02-08 by johanstokking