TheThingsNetwork/lorawan-stack #5183: Simulcast support for Class C multicast downlink

onizmx, 2022-02-06:
#### Summary

Simulcast support for Class C multicast device.
...

#### Why do we need this?

To cover some cases where we need to send exact same payload + absolute time to a single multicast device with a list of multiple gateways.
...

#### What is already there? What do you see now?

`scheduleDownlinkByPaths` returns upon a successful attempt while iterating over downlink paths [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/c52be1c2b65902568db228a15c7eae23b416a524/pkg/networkserver/downlink.go#L998)
We are currently sending a Class C multicast downlink to up to 8 gateways, Adding 1 second to `absolute_time` for each. This isn't very scalable for our business model.
A workaround for this is creating one multicast device per gateway (or a gateway cluster), which have the same app & network key as well as device address. We would prefer this to be the stack's feature if possible.
...

#### What is missing? What do you want to see?

Class C multicast downlink with `simulcast` or `use-simulcast` flag gets sent to all gateways included in `gateway_ids`, not just one.
...

#### Environment

3.17.2, but not related to a specific environment.
...

#### How do you propose to implement this?

Add `use_simulcast` boolean flag to ClassBC.
We already have a barebone / draft of the implementation which will be tested both locally & in our development environment.
A general guidance of the implementation will be much appreciated (especially how errors should be returned on a partial success).
Please have a look at the last 2 commits of this [forked branch](https://github.com/halter-corp/lorawan-stack/commits/feat/class-c-simulcast)
...

#### How do you propose to test this?

Test CLI with the newly added flag.
Check both multicast / gateway events to make sure downlink is sent to all gateways.
End device decoding.
Run the stack in our development / staging environment to make sure nothing breaks with or without the flag.
...

#### Can you do this yourself and submit a Pull Request?

Yes. But I would need a bit of guidance.
...

onizmx, 2022-02-08:
CC @adriansmares Maybe you are the best person to discuss about the feature?

virtualguy, 2022-02-14:
@johanstokking would love to get your opinion on this, it seems to technically work for gateways which are time sync'd with gps similar to class b beacons

adriansmares, 2022-02-14:
This is a feature that we're not in favor of right now, due to the fact that it encourages reusing the ABP session over a large amount of end devices _and_ over a large range. This is the definition of multicast, but this particular proposal is multicast².

My suggestion regarding your use case would be to use different ABP sessions in different locations in which you want to broadcast the downlinks. Keep in mind that you are open to replay attacks due to the fact that you are sharing the session over so much physical space, so you probably shouldn't have reused the session so much in the first place.

_In theory_, it could work if you can ensure that the gateways are always GPS synced - keep in mind that if you mix GPS synced and non GPS synced gateways, you may end up causing interference in the areas with overlapping coverage (as the same transmission will occur with different starting times).

I've taken a look at your fork and the only comment I have is that probably you don't want to propagate outside of `scheduleDownlinkByPaths` that you have scheduled multiple downlinks. Sure, you may have some extra events but there is no need to also propagate the scheduled downlinks - it's the same downlink transmitted multiple times. Other than that, it _probably_ should work, but I recommend that you reconsider if you really want to do this or not.

virtualguy, 2022-02-22:
Thanks for the feedback @adriansmares. We have unfortunately have situations where devices can move from coverage under one gateway to another within 15-60minutes which rules out different ABP sessions. The effort of resyncing the new session would cause substantially more traffic as the nodes move in large groups at a time.

I don't see how the increase of physical space exposes replay attacks. This is already an issue for devics that are out of coverage and could then have a message 'replayed' on them. I agree it widens the ability for an adversary to capture packets. In our use case we are only talking about a few km but I understand this feature could allow a multicast message to be sent across hundreds of gateways. The latter is already possible it just takes significantly longer to send.

The major use case for this feaure is for FWOTA for mobile devices. Standing up a session to send an upgrade and expecting the devices to be stationary for the duration of the FWOTA process is particularly challenging, especially for Class B devices which have lowered throughput

johanstokking, 2022-02-23:
Interesting discussion. I have a slightly different opinion than @adriansmares, I do think that this is useful.

Few thoughts:

1. Indeed in the early FUOTA designs and calculations for duty-cycle drain on a single gateway and large groups of devices not covered by a single gateway, it was suggested to use multiple gateways for the same multicast group to spread downlink capacity and expand the update area
2. Gateways are transparent in the LoRaWAN topology. Not only for uplink, but also for downlink. That means that multicast groups shouldn't be tied to a gateway
3. Class B beacons aren't authenticated. End devices should use class A `DeviceTimeReq` to obtain absolute time. There are some discussions in the Technical Committee to use authenticated beaconing within multicast groups. So end devices get a multicast group provisioned (via Remote Multicast Setup), and then the network can use that multicast group for authenticated beaconing

Like class B beacons are transmitted simultaneously by gateways at the same timestamp, authenticated beacons in class B multicast groups should also be transmitted simultaneously. I also understand the use case here that because of the transparency of gateways, you don't want to rely on the gateway-device physical link and let multiple gateways transmit the same class B application downlink frame in a multicast group.

Does Jamie's last comment and my thoughts change your opinion @adriansmares or do you have concerns?

adriansmares, 2022-02-23:
I cannot think of any extra concerns for now. As long as these are used for FUOTA and not broadcasting actuator commands, I do think that it can be useful.

onizmx, 2022-03-13:
@adriansmares @johanstokking Would use-simulcast boolean flag will suffice the API part?
And which part of the codebase will be the most suitable place for the actual implementation? It would be very helpful if we can get some guidance.

onizmx, 2022-03-13:
This was our draft api implementation https://github.com/halter-corp/lorawan-stack/commit/183b3e6a3599aedaf59dc6d9dae75ac69cc9fb27

johanstokking, 2022-03-15:
Thanks for the suggestions. We can't break the API nor can we change the default behavior. What we can do as an alternative to grouping the gateways in arrays (the one that breaks API), is to add a grouping field to `GatewayAntennaIdentifiers` so that NS/AS knows it's "one of all groups".

I'm also fine with a behavior flag on the `TxRequest` message.

So that leaves us with two options:
1. Add field to `ApplicationDownlink.ClassBC` that `gateways` means "all"
2. Copy `GatewayAntennaIdentifiers` to `ClassBCGatewayIdentifiers` and add nullable numeric `group` field. If null, follow current behavior. If set, one downlink path of all groups is attempted

What do you guys think?

onizmx, 2022-03-15:
@johanstokking The preferable solution for us will be 2, which is adding the `group` field. Just to clarify some cases:

1. [Two groups] `gtw_A (group: 0), gtw_B (group: 0)`, `gtw_C (group: 1), gtw_D (group: 1)` -> Will make at least 1 attempt per group. Therefore in this case, min attempts will be 2 (1 per group), and max attempts will be 4 (2 per group).

2. [Four groups] `gtw_A (group: 0)`, `gtw_B (group: 1)`, `gtw_C (group: 2)`, `gtw_D (group: 3)` -> Will make at least 1 attempt per group as well. Min & max attepts will be 4.

3. [Partial groups] `gtw_A (group: null)`, `gtw_B (group: null)`, `gtw_C (group: 0)`, `gtw_D (group: 1)` -> Will make at least 1 attempt per group. Min attempts will be 3 (1 for null, 2 for the groups), max attempts will be 4.

Because of the case 3, should we reserve 0 as default and replace when the `group` field is null?

johanstokking, 2022-03-16:
> * [Two groups] `gtw_A (group: 0), gtw_B (group: 0)`, `gtw_C (group: 1), gtw_D (group: 1)` -> Will make at least 1 attempt per group. Therefore in this case, min attempts will be 2 (1 per group), and max attempts will be 4 (2 per group).
> * [Four groups] `gtw_A (group: 0)`, `gtw_B (group: 1)`, `gtw_C (group: 2)`, `gtw_D (group: 3)` -> Will make at least 1 attempt per group as well. Min & max attepts will be 4.

Yes exactly.

> 3\. [Partial groups] `gtw_A (group: null), gtw_B (group: null)`, `gtw_C (group: 0)`, `gtw_D (group: 1)` -> Will make at least 1 attempt per group. Min attempts will be 3 (1 for null, 2 for the [...]

I was initially thinking to disallow mixing `null` group and numbered groups by returning `InvalidArgument`.

We can make `group` non-nullable and so `0` is then the default group. But our JSON API does currently not allow us to see the difference between not set and default value of non-nullable fields. So that would mean that `gtw_A (group: not set)`, `gtw_B (group: 1)` is effectively `gtw_A (group: 0)`, `gtw_B (group: 1)` and it would be processed as two groups. That might be confusing for end users. Therefore my idea was to make this explicit by making it nullable, and `group` must then either be set or not set for all gateways.

onizmx, 2022-03-20:
@johanstokking I agree about being explicit about the conditions. Thanks for the guidance!

closed 2022-06-08 by adriansmares