TheThingsNetwork/lorawan-stack #5197: Wrong URL format returned to LNS

onsolutionjames, 2022-02-09:
#### Summary
When using Basics Station LNS, the wrong URL can be provided to the gateway for MUXS. Basics Station seems to always expect the port to be specified, however TTS can return no port under some circumstances.

For example, if I set the `gcs.basic-station.default.lns-uri` or the gateway server address to `wss://stack.example.com:443`, TTS returns `wss://stack.example.com/traffic/eui-blabblah`, when it should have actually returned `wss://stack.example.com:443/traffic/eui-blabblah`.

#### Steps to Reproduce
1. Set `gcs.basic-station.default.lns-uri` to `wss://stack.example.com:443`, 
2. Follow the normal setup steps for a Gateway to connect it via Basics Station and CUPS, but make sure to leave the Gateway Server address field blank on the gateway
3. Basics Station tries to connect, but returns the error
`[TCE:ERRO] Illegal muxs URI (no port/path etc.): wss://example.com/traffic/eui-blahblah`

I'm not familiar with Go, but I have had a look through the code to see why the port isn't included. It seems to come from `info.Address`, which seems to come from this line: https://github.com/TheThingsNetwork/lorawan-stack/blob/95b5591a304e28e199d39a9754fc03777d1d7d17/pkg/gatewayserver/io/ws/ws.go#L128

Looking at the docs for Request.Host (https://pkg.go.dev/net/http#Request), it seems that this only returns the port _if the port is non-default for the scheme that was used_, so for example if you're connecting via `wss://` and you're connected via 443, it will omit the port, but if you use the scheme `wss://` but you're connected via 8887, the port will be included. The issue is that TTS currently assumes that it always returns the port.

(See [RFC 7230, section 5.4](https://rfc-editor.org/rfc/rfc7230.html#section-5.4): `if the connection's incoming TCP port number differs from the default port for the effective request URI's scheme, then a colon (":") and the incoming port number (in decimal form) are appended to the authority component`).

#### Environment
TTS v3.17.2 in Docker on a Ubuntu host (on EC2).

I have Application Load Balancer in front of the EC2 instance which I am using for SSL termination and re-routing LNS to port 443.

I am using a Dragino LPS8, running the latest firmware. 

#### How do you propose to implement this?
I don't have much experience with Go, but I've quickly made a fix here: https://github.com/jameseden1/lorawan-stack/commit/a151a2c6d82ccea07c9ebd8613626763c9f883de

I'm not sure if there is a better way to fix this, but I think this commit demonstrates the issue well. 

#### Can you do this yourself and submit a Pull Request?
If my linked commit above is an acceptable solution, just let me know and I can create a Pull Request

KrishnaIyer, 2022-03-15:
So the test would go to [`TestDiscover.go`](https://github.com/TheThingsNetwork/lorawan-stack/blob/c4b4ccc8e9778fe0cf540606ac404bbd57b7d8ab/pkg/gatewayserver/io/ws/ws_test.go#L217).

It's not the regex but we're checking that the port (80/443) is returned when a connection info request is made without a port.

onsolutionjames, 2022-03-19:
Hi @KrishnaIyer, it would be awesome if you could take over. I don't have much experience with Go and I'm not sure what the best way to write the test is

KrishnaIyer, 2022-03-22:
Ok Great. I'll pick this up :+1

PR #5330 (2022-03-23): Set default port if it doesn't exist in LBS discover message

#### Summary

Closes #5197 

The reasoning is in the issue
> Looking at the docs for Request.Host (https://pkg.go.dev/net/http#Request), it seems that this only returns the port if the port is non-default for the scheme that was used, so for example if you're connecting via wss:// and you're connected via 443, it will omit the port, but if you use the scheme wss:// but you're connected via 8887, the port will be included. The issue is that TTS currently assumes that it always returns the port.

#### Changes

- Use default port value [... cut]
closed 2022-03-23 by KrishnaIyer