TheThingsNetwork/lorawan-stack #5224: Scheduling downlinks via API seems inconsistent

nejraselimovic, 2022-02-17:
#### Summary
Scheduling a downlink (that contains the `decoded_payload` object) via API using  `/api/v3/as/applications/<app-id>/devices/<device-id>/down/push` requires `f_port` field to be defined, unlike when using `/api/v3/as/applications/<app-id>/webhooks/<webhook-id>/devices/<device-id>/down/push`.

#### Steps to Reproduce
Scheduling a downlink on `https://eu1.cloud.thethings.network/api/v3/as/applications/<app-id>/devices/<device-id>/down/push` using the `decoded_payload` object as explained [here](https://www.thethingsindustries.com/docs/getting-started/api/#schedule-downlink) doesn't work because it needs an `f_port` field defined, i.e. it produces the error mentioned below. When I add the `f_port` field to it, I see the downlink getting scheduled in the Console on the defined `f_port`.

On the other hand, scheduling a downlink on `/api/v3/as/applications/<app-id>/webhooks/<webhook-id>/devices/<device-id>/down/push` as explained [here](https://www.thethingsindustries.com/docs/integrations/webhooks/scheduling-downlinks/#scheduling-downlinks) using the `decoded_payload` object works without the need for an `f_port` field to be defined. As a result, I see the downlink being scheduled, but no `f_port` associated with it.

#### What do you want to see instead?
Scheduling downlinks using the `frm_payload` on both above mentioned URLs throws an error if there is no `f_port` defined, so I guess that scheduling a downlink without an `f_port` via `/api/v3/as/applications/<app-id>/webhooks/<webhook-id>/devices/<device-id>/down/push` should also throw an error.

Also, the error presentation in my terminal is different for these two scenarios. For  `/api/v3/as/applications/<app-id>/webhooks/<webhook-id>/devices/<device-id>/down/push` it looks like a nicely formatted JSON:

```json
{
	"code": 3,
	"message": "error:pkg/errors:validation (invalid `downlinks[0]`: embedded message failed validation)",
	"details": [
		{
			"@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
			"namespace": "pkg/errors",
			"name": "validation",
			"message_format": "invalid `{field}`: {reason}",
			"attributes": {
				"field": "downlinks[0]",
				"name": "DownlinkQueueRequestValidationError",
				"reason": "embedded message failed validation"
			},
			"correlation_id": "e147f3303106459eb1be87eb0b51d701",
			"cause": {
				"namespace": "pkg/errors",
				"name": "validation",
				"message_format": "invalid `{field}`: {reason}",
				"attributes": {
					"field": "f_port",
					"name": "ApplicationDownlinkValidationError",
					"reason": "value must be inside range [1, 255]"
				},
				"correlation_id": "a6042ac6841b4d8795b36e184ac1b3e1",
				"code": 3
			},
			"code": 3
		}
	]
}
```

while for `/api/v3/as/applications/<app-id>/devices/<device-id>/down/push` it's minified:

```json
{"code":3,"message":"error:pkg/errors:validation (invalid `downlinks[0]`: embedded message failed validation)","details":[{"@type":"type.googleapis.com/ttn.lorawan.v3.ErrorDetails","namespace":"pkg/errors","name":"validation","message_format":"invalid `{field}`: {reason}","attributes":{"field":"downlinks[0]","name":"DownlinkQueueRequestValidationError","reason":"embedded message failed validation"},"correlation_id":"c292a396be6244d895378dcbc0e4b9ed","cause":{"namespace":"pkg/errors","name":"validation","message_format":"invalid `{field}`: {reason}","attributes":{"field":"f_port","name":"ApplicationDownlinkValidationError","reason":"value must be inside range [1, 255]"},"correlation_id":"3822fa6ee76f410694256af55a85b424","code":3},"code":3}]}
```

It would probably be better for all errors to be shown as the one above, it's definitely easier to read. 

#### Environment
TTS CE on eu1

adriansmares, 2022-02-22:
So it seems that when we use the `AppAs.*DownlinkQueue`, the validation kicks in and rejects the request:

https://github.com/TheThingsNetwork/lorawan-stack/blob/78d2db0cb6b3e8458c44bea254acff82be5a2921/api/messages.proto#L267

While the AS webhooks do not go via any validation frontend:

https://github.com/TheThingsNetwork/lorawan-stack/blob/78d2db0cb6b3e8458c44bea254acff82be5a2921/pkg/applicationserver/io/web/webhooks.go#L390-L404

I guess we need the following:
- Validate the downlinks after the formatter unmarshals them in the web handler (and maybe MQTT)
- Allow `ApplicationDownlink` to have `FPort` 0 at API level, and reject the downlinks if the FPort is still 0 after the formatters ran.

ACK @johanstokking ?

johanstokking, 2022-02-23:
> * Validate the downlinks after the formatter unmarshals them in the web handler (and maybe MQTT)
> * Allow `ApplicationDownlink` to have `FPort` 0 at API level, and reject the downlinks if the FPort is still 0 after the formatters ran.

ACK

closed 2022-03-23 by adriansmares