TheThingsNetwork/lorawan-stack #5237: Unable to update multicast group

Oliv4945, 2022-02-22:
#### Summary
For multicast class B, new ping slot frequency, datarate and periodicity can not be changed from the GUI nor the CLI. What was first setup at multicast group creation seems to be kept internally.  
Same thing goes for multicast class C.

#### Steps to Reproduce

1. Setup a multicast group (B or C), use it
2. Change parameters on the GUI or the CLI
3. Those new parameters are correctly displayed in the GUI but the old ones are still used by the NS

CLI commands used:
```
ttn-lw-cli end-devices set 'd-test-app-1' 'multicast1' --mac-settings.rx2-data-rate-index 4
ttn-lw-cli end-devices set 'd-test-app-1' 'multicast1' --mac-settings.desired-rx2-data-rate-index 4
```

#### Environment
Tested on TTI v3.16.x and v3.17.2

adriansmares, 2022-02-22:
See https://www.thethingsindustries.com/docs/devices/mac-settings/#mac-settings-and-mac-state - if you update these settings via `mac-settings`, you will need to reset the MAC state of the end device. If you need to update them at runtime, you should update the `mac-state.desired-parameters` fields most likely.

Oliv4945, 2022-03-11:
Thanks @adriansmares, my colleagues confirmed that it resolved their issue

closed 2022-03-11 by Oliv4945