TheThingsNetwork/lorawan-stack #5269: Use Redis 7.0 features

adriansmares, 2022-02-28:
#### Summary

We should use Redis 7.0 features once 7.0 compatibility is widely available.

#### Why do we need this?

Certain operations can be more efficient this way.

#### What is already there? What do you see now?

`FindProtos` in `pkg/redis` uses `SORT`. `SORT` cannot be used against a read only replica, due to the potential `STORE` parameter (which we don't use).

#### What is missing? What do you want to see?

`FindProtos` should use `SORT_RO`.

#### Environment

`3.18.1`

#### How do you propose to implement this?

Update `SORT` command usage.

#### How do you propose to test this?

UT.

#### Can you do this yourself and submit a Pull Request?

Yes.

johanstokking, 2022-03-18:
We're a bit tied to the managed Redis offering from the major cloud vendors. They're generally a bit slow with upgrading Redis. Specifically Microsoft Azure is far behind.

Worst case, we have a backwards compatibility mode built in the Redis stores. For instance, we could take the read only client and use `SORT_RO` when the reported Redis version is 7.0 or higher, and otherwise the writer client and `SORT`?

adriansmares, 2022-03-18:
I've created this issue mostly for housekeeping. I know we won't have 7.0 for a long time.

Until we have 7.0 compatibility, I've written a read only compatible `SORT`-ish here - https://github.com/TheThingsIndustries/lorawan-stack/blob/26406f658dbbc29b03c3c8d2fe05ac650209e4aa/pkg/redis/redis.tti.go#L31 .

KrishnaIyer, 2025-04-30:
This can be closed.

closed 2025-04-30 by KrishnaIyer