TheThingsNetwork/lorawan-stack #5309: Browser-based CLI support in TTS

ymgupta, 2022-03-16:
#### Summary
Currently, TTS users have to download binaries/install `ttn-lw-cli`, configure it to their deployment, and then login/authenticate it before performing any operations.
We suggest implementing browser-based CLI support in the TTS  similar to AWS CloudShell.
#### Why do we need this?
To improve the user experience by eliminating the steps of installation, configuration, and authentication.
#### What is already there? What do you see now?
[ttn-lw-cli](https://www.thethingsindustries.com/docs/getting-started/cli/) tool.
#### What is missing? What do you want to see?
A Browser-based CLI in TTS similar to AWS CloudShell. Users should start a CLI session directly from the TTS console without the additional steps like installation, configuration, and authentication of the CLI.
#### Environment
TTS 3.18.1
#### How do you propose to implement this?
...
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No

KrishnaIyer, 2022-03-16:
I don't see the value in this. The console and the CLI are meant for different types of users. If your concern is that there are some actions that users really want to do via the console but can only do that now with the CLI, we can look into adding console support for that.

ymgupta, 2022-03-18:
We agree with you @KrishnaIyer in this regard, we will raise a feature request by identifying the gaps between CLI and console.

We are closing this issue now.

closed 2022-03-18 by ymgupta