TheThingsNetwork/lorawan-stack #5376: connect ETIMEDOUT 63.34.215.128:8333 in "MQTT Explorer"

jayjupdhig, 2022-04-11:
MQTT Explorer (http://mqtt-explorer.com/ , https://github.com/thomasnordquist/MQTT-Explorer ) screenshot HERE:

[image]

**Questions:**

Get that message when i try to log in: "connect ETIMEDOUT 63.34.215.128:8333"

1.) Why? Is the TCP daemon at endpoint 63.34.215.128:8333 not available?

2.) Do i need to select "Validate Certificate" in MQTT Explorer or not?

Screenshot of my MQTT Explorer Configuration here is attached...

Thank you very very much for your feedbacks... :-)

With best regards,
Jan

jayjupdhig, 2022-04-11:
Or is the encrypted connection currently not supported...?

KrishnaIyer, 2022-04-11:
The TLS port is `8883`

closed 2022-04-11 by KrishnaIyer

kosuodhmwa, 2022-04-11:
OH s**t, mea culpa - thank you very much!! :-)

"Validate Certificate" is also possible - because of that it's i.m.o. better to enable. :-) 

With best regards,
Jan