TheThingsNetwork/lorawan-stack #5394: Cache result of version check

htdvisser, 2022-04-19:
#### Summary

`ttn-lw-cli` should cache the result of the version check on disk, so that it doesn't have to do version check requests on _every_ call.

#### Why do we need this?

I'm working on a script that makes a batch of 100 devices claimable and generates QR codes for them. If the version check is enabled (which it is by default), I'm making 200 calls to the version check API, which is excessive.

#### Environment

ttn-lw-cli v3.18 on macOS

#### How do you propose to implement this?

Cache a successful version check response, together with a timestamp, in `os.UserCacheDir`. Only do a version check (1) if that file does not exists or (2) if the timestamp is too old.

htdvisser, 2022-04-19:
CC @NicolasMrad for triage: this needs a milestone.

michalborkowski96, 2022-04-21:
Why can't you just pass the `--skip-version-check` option?

htdvisser, 2022-04-21:
Of course I can update my config to disable the version check, but I _do_ want to be notified about new versions. I just think the CLI shouldn't run the check 200 times in 10 seconds.

michalborkowski96, 2022-04-21:
Yeah, you're right. I'll start working on this once I get through more important issues

PR #5415 (2022-05-16): cli: Use cache for version check
#### Summary
Closes #5394 

#### Changes
Made version check use cache

#### Testing
1. Run one time - information about version check. Run second time - no information.
2. Edit the cache file to an early date and run - version check gets performed.
3. Edit the cache file to make it incorrect and run - warning shows up, then version check gets performed, subsequent run doesn't show version check information.

##### Regressions
None

#### Notes for Reviewers
Maybe instead of coding the cache logic in `pkg/version/remote.go`, it'd be better to create an all-purpos [... cut]
closed 2022-05-16 by michalborkowski96