TheThingsNetwork/lorawan-stack #5397: zsh autocompletion ttn-lw-stack.ttn-lw-cli

flapotre, 2022-04-20:
#### Summary

After following the documentation to install ttn-lw-stack on ubuntu 20.04 using snap to use the cli, I was expecting to have auto-completion on ZSH console as [documentation](https://www.thethingsindustries.com/docs/getting-started/cli/installing-cli/) sugest it. Unfortunately, only bash completion work.

I did follow [this documentation](https://www.thethingsindustries.com/docs/ttn-lw-cli/ttn-lw-cli_completion_zsh/) that explain how to set auto-completion by hand for zsh.

I test it on VirtualBox with latest Ubuntu LTS installed to be sure that my own zsh configuration was not evolve in this issue. 

#### Steps to Reproduce

##### Create a clean environment (optional)
I could not run snapd in docker and didn't have time to start investigating why so I use virtualbox to have clean env: 
[Download Ubuntu](https://www.ubuntu-fr.org/download/)
[Install VirtualBox](https://linuxize.com/post/how-to-install-virtualbox-on-ubuntu-20-04/) 
[Install Ubuntu on virtual box](https://brb.nci.nih.gov/seqtools/installUbuntu.html) 

##### Mandatory steps

1. Update and upgrade your system `sudo apt update && sudo apt upgrade -y`
2. Install ttn-lw-stack : `sudo snap install ttn-lw-stack`
3. Enable zsh completion : ` echo “autoload -U compinit; compinit” » ~/.zshrc`
4. launch zsh REPL `zsh`
5. Try to get completion: `ttn-lw-stack.ttn-lw-cli us<TAB>`

You also can try to set completion by hand, by following [documentation ](https://www.thethingsindustries.com/docs/ttn-lw-cli/ttn-lw-cli_completion_zsh/) you have the same result 

6. Generate Zsh completion script : ttn-lw-stack.ttn-lw-cli completion zsh > “${fpath[1]}/_ttn-lw-cli
7. Restart zsh Terminal `<Ctrl> + <Alt> + t`
8. Try to get completion: `ttn-lw-stack.ttn-lw-cli us<TAB>`

#### What do you see now?

nothing is happening

#### What do you want to see instead?

I would expect to get prompte to chose between `use` or `users` as it does on 

#### Environment

Already described 

#### How do you propose to implement this?

I don't know what wrong 

#### How do you propose to test this?

I don't know what wrong 

#### Can you do this yourself and submit a Pull Request?

I don't know what wrong

nicholaspcr, 2022-06-16:
Hi @flapotre, my apologies for the delayed response on this issue.

At first I had the same problem when trying to replicate the steps and marked this as bug, after paying more attention I realized that the command given by the documentation `ttn-lw-cli completion zsh` in fact generates a bash autocomplete script due to the lack of the `--shell` flag before `zsh`. 

After adding the flag I couldn't replicate your issue on a vm with Ubuntu. I would recommend trying to run the autocomplete setup with the command `ttn-lw-cli completion --shell zsh` and see if that solves your issue.

nicholaspcr, 2022-06-20:
Considering that the documentation PR has been added to correct the lack of the `--shell` flag in the example, I'll be closing this PR.

@flapotre If you still have any problems with the autocomplete feel free to write about it here.

closed 2022-06-20 by nicholaspcr