TheThingsNetwork/lorawan-stack #5453: Missing field in MQTT json or

pe1mew, 2022-05-13:
#### Summary
frm_payload is missing in MQTT json struct
...

#### Steps to Reproduce
Monitor MQTT data with NodeRed.

#### What do you see now?
In the json struct that is published over MQTT,  the field ```msg.payload.uplink_message.frm_payload``` is missing. 
It seems to be replaced with ```msg.payload.uplink_message.session_key_id``` the content is not a HEX number but seems base64 data: ```AYBH/WpoSmh/h0iI1kb39g==````.

[image]

#### Environment
MQTT node in NodeRed.
devID 58A0CB0000202E3A
appEUI 58A0CB0001500000

#### How do you propose to implement this?
Correct the error and reinsert the base64 encoder payload.

#### Can you do this yourself and submit a Pull Request?
NO

closed 2022-05-17 by johanstokking

pe1mew, 2022-05-17:
I'm afraid I have to disagree with the assumption that this is an empty uplink message on FPort 8. 
Although it was not shown in the JSON struct, the payload was decoded correctly and populated with the information sent by the device. 

I suggest investigating the issue further.

johanstokking, 2022-05-18:
Hmm, please paste the anonymized full JSON uplink message so we can investigate, and reopen the issue, thanks.