TheThingsNetwork/lorawan-stack #5456: Support additional fields for importing end devices from CSV file format.

ymgupta, 2022-05-14:
#### Summary
The Things Stack currently supports importing end devices from CSV (comma-separated values) files with limited device fields. Users are seeking support for additional columns (e.g., device class and Rx2 Data Rate, etc).
#### Why do we need this?
Adding the ability to import devices via CSV with additional fields, similarly to the JSON format for end devices, will eliminate the need for separate configuration of other fields via CLI/API, etc.
#### What is already there? What do you see now?
We can currently import an end device with the few fields specified in the [CSV File Reference](https://www.thethingsindustries.com/docs/getting-started/migrating/device-csv/).
#### What is missing? What do you want to see?
Importing devices with more specific fields similar to the [JSON End Device Format](https://www.thethingsindustries.com/docs/getting-started/migrating/device-json/#json-end-device-format).
#### Environment
TTS v3.19.1
#### How do you propose to implement this?
Add support for additional fields in the CSV import feature.
#### How do you propose to test this?
...
#### Can you do this yourself and submit a Pull Request?
No

johanstokking, 2022-05-17:
What if we load the device profile from the device repository?

What is the scenario for allowing all fields, and not using JSON for import?

ymgupta, 2022-05-18:
In general, users are preferring the CSV import as they get the device details from the manufacturers as CSV files. So, they want to make a few additions to the CSV file and import it to avoid any further updates with API/CLI (`supports_class_c`/`supports_join` & Session Keys for ABP devices, etc).

Therefore, it would be good to add support for the most frequently used fields (for example, fields that we allow to modify from TTS Console) to the CSV import feature.

johanstokking, 2022-05-18:
OK, I see.

kschiffer, 2022-06-02:
> What if we load the device profile from the device repository?

+1 on this -> #5512

nicholaspcr, 2022-06-02:
@kschiffer Looks like a good idea to me, I can take a look into doing that on the backend. 
Like you said on #5512, if it's on the backend it would also be applicable to the CLI. So it's probably best.

johanstokking, 2022-06-05:
Yes I'm a big fan of #5512. We should do that anyway.

Still, we should support this too. It's not too big of a deal to add additional fields support in the CSV import, and I'm sure this can come in handy.

PR #5582 (2022-07-08): Add additional fields for csv converter

#### Summary

Closes #5456.

While loading the device profile from the DR solves a lot of the problems in devices migration, there is interest in having at least some of the most common [JSON](https://www.thethingsindustries.com/docs/getting-started/migrating/device-json/#json-end-device-format) fields settable via CSV.

#### Changes

- Added more csv field setters.
- Updated unit tests in `devicetemplates` pkg.

#### Testing

Unit testing and a bit of 🐒 

##### Regressions

There shouldn't be one as this is just adding more fields in which are accepted [... cut]
closed 2022-07-08 by nicholaspcr