TheThingsNetwork/lorawan-stack #5458: Add support for reading notifications in the CLI

htdvisser, 2022-05-16:
#### Summary

As a follow-up of https://github.com/TheThingsNetwork/lorawan-stack/issues/4935 we should add Console support for reading notifications and marking them as seen.

#### Why do we need this?

So that users can also see and manage notifications for which The Things Stack doesn't send emails.

#### What is already there? What do you see now?

API: `GET /users/{user_id}/notifications` to list notifications and `PATCH /users/{user_id}/notifications` to update the status of notifications. See `notification_service.proto` for more details.

#### What is missing? What do you want to see?

@kschiffer should probably come up with the UI/UX for this, but I'd like to see something similar to GitHub notifications:

- A "bell" icon with an indicator for unread notifications
- A table view of all (not archived) notifications, and on each row:
	- Indicator if status is unread
	- Checkbox for bulk update (status can be updated to unseen / seen / archived)
	- Icon for entity type
	- Entity ID
	- Notification type (until we can sync the "registry" with the frontend, I think we should just use the raw text here)
	- Time (ago)
	- Some way for the user to get the details (the full JSON) of the notification

#### Can you do this yourself and submit a Pull Request?

No. Let's start with some UI/UX ideas from @kschiffer.

adriansmares, 2022-08-16:
I've moved this to `3.22.0`. While the implementation itself does not have to land in that release, please consider designing the UX for that release.

KrishnaIyer, 2023-11-09:
Phase 3 of https://github.com/TheThingsIndustries/product-management/issues/29

PR #6479 (2024-02-07): Support notifications in the Console

#### Summary

Closes #5458 

#### Changes

- Add notifications view
- Fetch notifications every 5 minutes
- Add more utility classes

<details>
<summary>Notifications screenshots</summary>
[image]
[image]
[image]
</details>
<details>
<summary>Archive screenshots</summary>
[image]
</details>
<details>
<summary>Responsive screenshots</summary>
<img width="391" alt="Screenshot 2023-08-23 at 11 30 34" src="https://github.com/TheThingsNetwork/lorawan-stack/assets/56658938/3a42ea45-b071-4a84-84d3-e0e
closed 2024-09-02 by KrishnaIyer